cortex-lab/KiloSort #11: automatic setting of number of channels

petersaj, 2016-04-28:
Marius' note: 

It needs to know how many total channels there are so it loads them correctly from the bin file. In the past I left this as a user-defined option because there were cases where we ran it without a prb file.... But I think these days the prb file is a requirement for loading into Phy, so it needs to be made for any type of probe anyway. So I should really just determine the number of channels from the prb file.

nsteinme, 2016-04-28:
Another option here is to make the input data file type be an NPY file
rather than flat binary. This file type contains a tiny header which
specifies the data size and type. It can be memory mapped and "fread" (by
just skipping the header first). I believe an existing data file can be
converted to NPY by concatenating a file of the header with the data file;
I need to test this more.

So that option is nicer because it is self-contained - if we want to go
that way it would be best to do it in coordination with Cyrille and phy so
you can use the same raw data file for both. Simply using the same
params.py file that phy uses is definitely the simpler option.

marius10p, 2016-04-28:
Thanks Nick, but just making the NPY file out of a binary might take a significant amount of time. On top of that, memory mapping in matlab works significantly worse than just reading a flat binary file, I have noticed recently.

nsteinme, 2016-04-28:
In linux the concatenation is immediate. But I'm struggling to get it to
work on windows, so we can forget it for now.

Fwiw, you can of course still just read the npy file with fread in matlab,
you just have to skip the header (usually 80 bytes) first.

marius10p, 2016-09-17:
The number of active and total channels is now read from the channel map file, unless already specified in ops. Same goes for sampling frequency.

closed 2016-09-17 by marius10p