cortex-lab/KiloSort #12: Only uses gpuDevice(1)

theilmbh, 2016-07-20:
In @gentnerlab we are attempting to run KiloSort on a server with multiple GPUs.
We have to ration the use of each GPU as multiple lab members use the different GPUs at the same time. 

It appears that KiloSort overrides any previous gpuDevice setting by explicitly calling gpuDevice(1).  We'd like for there to be a setting to tell KiloSort which gpuDevice it should be using.

marius10p, 2016-07-22:
I have over-written this. gpuDevice now only gets called in the master file, so you are free to use it as you want. This is because GTX 10xx are not yet supported by Matlab and seem to recompile source for a few minutes everytime gpuDevice(1) is called.

closed 2016-07-22 by marius10p

MarvinT, 2016-07-27:
It seems that there are still a few calls of gpuDevice(1) in some of the run_reg_mu functions. are these not used or should we not be using the master branch?

nsteinme, 2016-07-29:
Yeah, sorry, there are many old functions remaining in the repository which have been superseded by newer versions (and some may have calls to gpuDevice(1)). 

I have kept them around for a while, just to have them close at hand, but I will remove un-necessary files in an update, not too far in the future.

marius10p, 2016-07-29:
Yeah, sorry, there are many old functions remaining in the repository which have been superseded by newer versions (and some may have calls to gpuDevice(1)).

I have kept them around for a while, just to have them close at hand, but I will remove un-necessary files in an update, not too far in the future.