cortex-lab/KiloSort #13: Spike detected twice?

ablot, 2016-07-21:
Is it possible that the same spike is matched by 2 clusters, therefore duplicating it? I have a few clusters that look like that (see screenshot). They have similar waveforms and crosscorrelograms plus a crosscorrelogram with a double peak at +/- 0.4 ms  (bins are 0.2ms on the screenshot)
[image]

The slightly annoying point is that it seems to be only partially a duplicate, so I cannot just put one of the cluster to noise and move on with the other. Should I just reduce the number of cluster to detect (`ops.NFILT`) to try to avoid that?

nsteinme, 2016-07-21:
Can you show us what the template waveforms themselves look like (press "w"
in phy)?

I have seen very rarely that spikes from one template will get fitted to
the residual of another - but in that case the CCG shows a peak only 1 bin
wide, not a few ms wide like you show. I think it is possible instead that
you are looking at a real feature of the data? We can see in the waveform
samples that there's an extremely high rate of spikes on those channels,
and the clusters you're showing aren't too much higher than the noise
level, plus have a fairly generic shape...?

Nick

ablot, 2016-07-21:
Yes you're right, it is in the data.

Here is the template: 
[image]

And an overlay of 200 spikes (raw trace, baseline subtracted) of cluster 95 that are less than 1ms apart from  a spike of cluster 0 (average in red):
[image]

There are really multiple events separated by less than a ms. I guess there is no magic, when there is lot of activity, kilosort cannot somehow isolate small spikes out of the mua cloud.

closed 2016-07-21 by ablot