cortex-lab/KiloSort #20: post hoc merge fails

MarvinT, 2016-07-29:
I know the example master file says that posthoc merge templates is still under construction however I was hoping I could get some help getting it working or an explanation on where to start a pull request to fix it.

The error I'm getting is:
```
Error using <
Matrix dimensions must agree.

Error in merging_score (line 22)
steps = sum(hs1(1:m1)<mmax/troughToPeakRatio) + ...

Error in merge_posthoc2 (line 89)
            mo = merging_score(f1old - f2old, f1new-f2new, ops.fracse);

Error in master (line 112)
rez = merge_posthoc2(rez);
```

and seems to be caused by l1 being positive in merging_score causing b2 to be empty and everything to fail. I don't have a good enough sense on what the variables mean to know what should be happening here.

marius10p, 2016-07-30:
Hi Marvin, 

I am not sure why you are getting that error. 

I have not been updating that piece of code together with the rest. The reason is, we are now testing a few others ways to reduce the number of manual merges, like drift tracking. Once those methods do their job, there will be less work to do for this "posthoc" merging step, and it will probably need different assumptions, if drift is already taken care of. 

So, a new version of Kilosort will deal with non-stationarity, but that may take between a few weeks and a few months. If I have time in the next week or two, I will try to ensure posthoc_merge works in the mean time.

marius10p, 2016-09-17:
Hi again,

I think I found the bug with this step and it's fixed now. If you run rez = merge_posthoc2(rez),  and then export to Phy using rezToPhy(rez, fpath), it will export the clusters merged by the final step. Soon we will have an option in Phy to undo this kind of merge, if the user thinks it was wrong. 

There is also a fast, complete working example now in /eMouse where you can test the merge_posthoc2 effect.

closed 2016-09-17 by marius10p