cortex-lab/KiloSort #21: Indexing of numpy output is unclear

gvoysey, 2016-08-02:
@rswilliamson and I have been using KiloSort and Phy for spike sorting over at @Polley-Lab. 

I have been working from [this](http://data.cortexlab.net/dualPhase3/data/script_dualPhase3.m) example's comments to correctly align templates and spike times to concatenated raw data.

Direct examination of the `spike_templates.npy` and `spike_clusters.npy` arrays in python suggest that, contrary to the comment in the above file, they store as values 1-indexed indicies: 

```python
In [30]: spike_templates = np.load('spike_templates.npy')
In [34]: spike_templates.min()
Out[34]: 1
In [35]: spike_templates.max()
Out[35]: 127

In [36]: spike_clusters_clu = np.load('spike_clusters.npy')
In [37]: spike_clusters_clu.min()
Out[37]: 1
In [38]: spike_clusters_clu.max()
Out[38]: 167
```

If this is the case, then this code is probably inconsistent, because `clu` and `ss` are unchanged from their numpy representation but `spikeTemplates` has 1 added to all its values, and doing both of those things at the same time doesn't seem correct. 

```matlab
%% Load in the results from Phy and plot statistics about the good clusters
    clu = readNPY(fullfile(penDir,  'spike_clusters.npy'));
    ss = readNPY(fullfile(penDir,  'spike_times.npy'));

    load(fullfile(penDir,'rez.mat')); %the raw templates are in rez.Wraw
    raw_templates = rez.Wraw;
    clear rez;

    spikeTemplates = readNPY(fullfile(penDir,  'spike_templates.npy')) + 1; % note: zero-indexed

    [cids, cgs] = readClusterGroupsCSV(fullfile(penDir,  'cluster_groups.csv'));

    %% Plot the good clusters with their filtered raw data and templates.
    goodClusters = cids(cgs==2);
```

Confusingly, we get reasonable-looking output when plotting, for each cluster, the template(s) that was/were used versus the averaged data across each channel. 

Also confusing, the output of [this](https://github.com/cortex-lab/sortingQuality/blob/a7c7f29eb64e6dc952c676643e9195502397b1f8/core/maskedClusterQualitySparse.m) function, when assessing cluster quality and separation index, does not list a clusterID of 0 either. 

I might be over-thinking this, but i'm now confused about whether or not there's a fencepost error in my processing code.

nsteinme, 2016-08-02:
Are you sure that Kilosort simply did not return any spikes for template
number zero? If there is a template fit in the algorithm that, in the final
step, gets zero spikes assigned to it, then there aren't going to be any
entries in spikeTemplates with that template's number. Usually this seems
to happen for about 5% of template numbers for me (but will depend a lot on
your data and kilosort parameters). So for instance you might see that
other numbers - besides zero, but less than 127 - are also not represented
in spikeTemplates. Let me know whether or not that may be a possible
explanation for what you're seeing? The most direct test here is to load
the raw data in phy, see the shape of the waveforms/template for some
random template (template 37, say) then see whether that matches
templates(37,:,:)
or templates(38,:,:) in the data you've loaded into matlab (should be the
latter) and/or see whether the raw waveforms you extract from the spike
times corresponding to spikeTimes(spikeTemplates==37) or
spikeTimes(spikeTemplates==38) match that template shape (again, should be
the latter).

gvoysey, 2016-08-05:
in short, i'm an idiot and fenceposted wrong.  All's well.

closed 2016-08-05 by gvoysey