cortex-lab/KiloSort #22: Kilosort fails for nchan=1

rswilliamson, 2016-08-10:
I've been trying to run kilosort on some single channel data. 
FitTemplates line 43 fails because squeezing with two singleton dimensions yields a matrix with differently ordered dimensions. 
There may also be some additional nchan=1 bugs (because trying to fix the squeeze issue myself resulted in further downstream issues).

nsteinme, 2016-08-10:
Suggestion for a quick workaround: put a second row in your data file full
of zeros (or maybe random values would be less likely to provoke further
crashes) and call it a two-channel recording.

kkatlowitz, 2016-12-28:
this solved it for me, but I had to do three channels because Nrank=3 in optimizePeaks.