cortex-lab/KiloSort #24: Handle multiple files

m-okun, 2016-08-22:
It would be really useful if several files can be dealt with as one continuous recording without the need to concatenate the files.

mspacek, 2016-11-07:
It would also be nice if gaps between recordings were take into account, especially when considering drift. This would require the user to somehow provide a global time stamp (or global datetime stamp) for the start (and maybe end) of each recording relative to some t=0, such as the start of the first recording.

marius10p, 2016-11-19:
Hi Martin, can you tell me a bit more about why the exact duration of gaps is important? Drift tracking is not yet integrated into Kilosort, but when it will be, it won't really care about the gap duration between blocks. Let's talk more when I introduce the drift tracking module.

mspacek, 2016-11-23:
Hi Marius. Sorry for the delay. The issue is that if there are gaps between recordings, and if the duration of those gaps is unknown during spike sorting, then drifting units will appear to suddenly jump from one position to the next. That nonlinearity is harder to understand than if gap durations are accounted for. Then at least you can linearly interpolate from one side of the gap to the other, and assume constant drift during that time. But as you wrote in some other comment, best of all would be gapless recordings.