cortex-lab/KiloSort #26: Hi,

hendrikrth, 2016-09-08:


nsteinme, 2016-09-08:
Hi, great to hear you are having success with Kilosort. The channels are displayed in phy template-gui according to the xcoords and ycoords variables so you just need to flip the ycoords to invert the depth, the simplest thing would be `ycoords=-ycoords;`. You can avoid re-running Kilosort if you just change channel_positions.npy directly. You should find that channel_positions.npy has two columns, the first of which is xcoords and the second is ycoords - so just make a new channel_positions.npy with the updated values, then re-open phy template-gui.

hendrikrth, 2016-09-12:
I changed the channel_position.npy the way you explained it. This altered the arrangement of the channels in the waveform view. But the order of the traces in the traceview was not changed. Also the depth in the clusterview were not altered. I then rerun kilosort again with the changed parameters. The depth in the clusterview now fits. It seems to me that the arrangement of the channels in the traceview and in the waveform view starts with channel 0 at the bottom and then increase upwards. But this shouldn't be a to big problem, because it is just a visualisation issue I can work with. But I still have another question about the adjacency graph (the kcoords variable, if I understand it correctly). So far it is a vector of ones of the length of channels. My probe has 16 channels, which alternate from left to right. So first there is channel 0 on the left, then comes channel 1 on the right, then channel 2 on the left again. So I wonder if my kcoords variable only connects channel 0 to channel 1 and not to channel 2. Is this assumption true, or am I missing something?

marius10p, 2016-09-12:
Hi, there is not adjacency map in Kilosort. Spikes are given a "mask" by their low-dimensional representation, which is faster and more robust than the KK-style masks. 

The kcoords variable is for completely isolating the processing of separate shanks / tetrodes. If you channels are next to each other, you don't have to worry about that.

marius10p, 2016-09-17:
hi again, there was an error with the way Kilosort was exporting the channel maps to Phy. It is fixed now. There is also a complete working example in /eMouse (including a made-up channel map) which you can use as a template and export to Phy.

closed 2016-09-17 by marius10p