cortex-lab/KiloSort #27: Spike Window

kkatlowitz, 2016-09-15:
Would it be possible to make the number of samples to take around each spike into a parameter? Currently it is set to 61 - sampling at 20kHz makes it into about 3ms.

marius10p, 2016-09-17:
The number of samples was very much hard-coded into everything. Took a bit of work but I fixed all of those. You can now specify the number of samples with ops.nt0 and I tested that it works on the new simulated data in /eMouse. This can be used in general as a very fast complete working example.  

For now ops.nt0 is defaulting to 61 and hidden from the configuration file. Let me know if you find errors, and I will eventually make it an "official" option in the configuration file.

closed 2016-09-17 by marius10p

samReiter, 2016-09-28:
Thank you very much for adding this. If I increase the value from 61 I run into errors with the hardcoded values for removing peaks in isolated_peaks.m and for setting the spike peak time in get_PCproj.m when I initialize from data. What do you think about having these values adjust with obj.nt0? I would do a pull request but I run into an error with CUDA if I try to make the samples per spike over a certain size. 
I tried to set ops.nt0 = 97, dt in get_PCproj.m to 41, and in isolated_peaks.m I made line 10 
peaks([1:40 end-80:end], :) = 0;

No error if I keep ops.nt0 in the 80s, not sure where the problem is.