cortex-lab/KiloSort #31: eMouse cannot be run without a GPU

yasuiniko, 2016-10-09:
**Description**
I'm a bit new to MATLAB, so I apologize if this issue is due to a lack of experience. However, if that's the case it might be nice to expand the documentation to cover this issue.

In `master_eMouse.m`, there's a `useGPU` flag, works to avoid using the GPU when making the data. However, the flag doesn't prevent `optimizePeaks.m` from being run, which does require a GPU. Is there a similar function that doesn't require a GPU that could be swapped for `optimizePeaks`, or is the program meant to be used on a device with a GPU?

I saw that `ops.initialize` could be used to prevent `optimizePeaks` from being run, but based on the comments it seemed like that would prevent the data from `make_eMouseData` from being used in the analysis.

Any insight you can offer would be greatly appreciated.

**Error Message**
```
There is a problem with the CUDA driver or with this GPU device. Be sure that you have a
supported GPU and that the latest driver is installed.

Error in optimizePeaks (line 51)
U = gpuArray(uBase(ind_filt, :))';

Error in fitTemplates (line 32)
        WUinit = optimizePeaks(ops,uproj);%does a scaled kmeans

Error in master_eMouse (line 25)
rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively

Caused by:
    The CUDA driver could not be loaded. The library name used was
    '/usr/local/cuda/lib/libcuda.dylib'. The error was:
    dlopen(/usr/local/cuda/lib/libcuda.dylib, 10): image not found
```

marius10p, 2016-10-09:
The easiest thing to do is set ops.initialize  = 'no', and it will run fine, though it might produce slightly suboptimal results. 

I will take a look at the function also, and make sure it can run on CPUs.

micheleacox, 2016-10-26:
@marius10p, I am also looking to use this without GPU initially (have to demonstrate usefulness to research group before investing time/money into stating GPU computing, we're not using it at the moment). Thanks for the tip on the ops param!

marius10p, 2016-10-26:
@micheleacox You should be able to grab a 4GB GTX 1050Ti for $150 soon. Will be 2-3x times slower than the best GPU, but still 10-20x times faster than the CPU version! 

Let me know if there's any way I can help, I'll try to get to the CPU initialization option soon.

nikolaskaralis, 2017-03-17:
Hi @marius10p,
Do you have any suggestions on the best GPU to use with KiloSort?

Thanks in advance.

Nikolas

marius10p, 2017-03-17:
Hi @nikolaskaralis 

Any Nvidia GTX 10xx GPU will be good enough. If you want it to run faster, the speed will pretty much scale with price. Best, - Marius

nikolaskaralis, 2017-03-17:
@marius10p  Thanks for the info.
Btw, optimizePeaks indeed has no option for no-GPU case so one need to run without initialization.

nikolaskaralis, 2017-03-18:
Hi @marius10p
I am considering the GTX 1080 but I see there are compatibility issues with even recent Matlab versions. I see somewhere in the code that this is the one you probably use. Could you please share what is a working combination in your experience? (GPU, OS, Matlab version)

Cheers,
Nikolas

kkatlowitz, 2017-03-20:
I am using kilosort with GTX 1080, Windows 10, Matlab 2016a, CUDA 7.5

nikolaskaralis, 2017-03-20:
Great, thanks for this info!

marius10p, 2017-03-20:
We use Kilosort with GTX 1080, Windows 10 or Linux, Matlab 2016b and CUDA 7.5.

nsteinme, 2017-03-20:
I've used the gtx1080 and matlab2016a, but with windows 7, which also
works.