cortex-lab/KiloSort #32: Redefining principal components

SheinIdelsonMark, 2016-10-24:
Hi

I was wondering if you have a script for calculating the predefined PCs. You mentioned in the config file that these should be recomputed for new data sets. Is this just the weight matrix for a sample of representative spikes? Are the PCs computed on raw data or filtered data (which filter)?

Many many thanks,
Mark

marius10p, 2016-10-24:
Hi Mark,

Yes, these are just the PCs of a sample of representative spikes (just their largest channels), aligned at sample 20. They are the temporal PCs, not the spatial masks. You can see it's just a .mat file that is loaded from disk. 

I should really add the PC computation to the preprocessing: will do so at the next update. It really should not play much of a role in the sorting itself, but it might make the PC visualizations in Phy clearer. 

Best,
   Marius

SheinIdelsonMark, 2016-10-25:
Hi Marius,

Thanks for the info. If this will not create major issues in Phy, I will
just wait for your next update.

Many thanks and all the best,
Mark

-- 
Mark Shein-Idelson (PhD)
Max-Planck-Institute for Brain Research
Department of Neural Systems & Coding
Max-von-Laue-Str. 4
60438 Frankfurt am Main
Germany

email: Mark.Shein-Idelson@brain.mpg.de
institute homepage: www.brain.mpg.de
personal homepage: https://sites.google.com/site/marksheinidelsonwebsite/
phone: +49 - 69 - 85 00 33 - 2030

LauriTK, 2017-12-20:
Hi,

I am interested also on this question.  How I can compute my own PCs? I'm getting some visual bugs in Phy. For example can't see waveforms at all, only black window with channel numbers (in wrong arrangement). There is also problems with the TraceView. Had no problems with eMouse data. Is this because I used your PC-file or is this something that is only Phys 'feature'?

Thank you for help and for amazing tool!

closed 2025-02-09 by SheinIdelsonMark