cortex-lab/KiloSort #39: Simplify mexcuda compilation on linux systems?

ereming, 2016-11-30:
I noticed that I can compile the CUDA files by removing the .xml files and changing `mex *` to `mexcuda *` in lines 9-11 of mexGPUall.m. eMouse ran fine afterwards. Running Matlab R2016b on a linux cluster w/ CUDA 7.5 loaded up.

marius10p, 2016-11-30:
Done. Thanks for noticing that. We had noticed it works on Windows. Good to hear Matlab is simplifying this process on all platforms. 

Are you sure it is necessary to remove the .xml files? I thought mexcuda automatically looks in the Matlab install location first. I like to keep them there as examples, if the easy install fails for whatever reason.

ereming, 2016-11-30:
No, it's fine without removing them.

closed 2016-11-30 by marius10p