cortex-lab/KiloSort #41: Force removal of template?

ereming, 2016-12-10:
I'm interested in experimenting with something. Would it be straightforward to kill a cluster manually, then re-run part of the sorting algorithm to resort that cluster's spikes among the remaining templates? One special case would be where I suspect that a particular cluster may actually be several neurons firing together (also see https://github.com/cortex-lab/KiloSort/issues/10). Would it be possible to remove the template for that neuron manually, then rerun the step which determines overlapping spikes (from the remaining templates)?

marius10p, 2016-12-10:
You can do this by setting to zeros the corresponding array of rez.dWU(:,:,k) = 0 for the cluster k you want to remove. Then run fullMPMU again. Keep in mind that Phy uses zero indexing.

nsteinme, 2016-12-10:
If you do try this, I'd be really happy to know how it goes - the question
is whether all of those doubled-spikes really do get assigned to both of
their "true" clusters or whether something else happens to them. It's a
great idea of a thing to try - let us know how it works out!

Nick

ereming, 2018-03-27:
I have a question about this technique, related to how KiloSort implements thresholding for spike detection. Is the threshold applied to single channels, or groups? Will an event with low energy on multiple channels get counted as a spike, but not an event with the same amplitude on a single channel? I ask because I'm sorting a file and noticing that there are several clusters with the following properties:

1. spatially segregated peaks.
2. a moderate number of spikes - too many for them to be statistically likely to result from the temporal collision of a pair of any of the other clusters.
3. template shapes on the individual electrodes which are much smaller than any of the other clusters to which they might otherwise be assigned.

The curious thing is 3 - I don't see clusters with the same (small) amplitudes but which exist only at a single location (i.e. fewer channels).

I think the solution is probably to just ignore these clusters as noise, so I'm asking mostly out of curiosity. It wasn't clear to me by going back through the paper.