cortex-lab/KiloSort #44: cpuMPmuFEAT returns empty sts

benballintyn2, 2016-12-16:
Hello,
I am trying to analyze 128 channel data using only a cpu at the moment (useGPU = 0, ops.initialize = 'no'). I am getting an index exceeds matrix dimensions error in fullMPMU (line 204) because the variable st3 is empty. This seems to be because the function cpuMPmuFEAT is returning an empty st variable but I cannot figure out why. Any help you could provide would be greatly appreciated.

marius10p, 2016-12-16:
Hi, 

This implies that Kilosort did not detect any spikes at all. Does it find sensible templates during fitTemplates.m, based on the plotting you see and the rez.Wraw?  

Try lowering the values in ops.Th. Also take a look at the filtered and channel-whitened data ("DATA" in your workspace), make sure it looks sensible. If the channel map is not quite right or the number of channels is off, then it could have jumbled together dimensions.

micheleacox, 2016-12-21:
I run into a similar issue only when using the "fromData" initialization option (so not otherwise). 

Here is my command line output and error:

```
Time   0s. Loading raw data... 
Time   7s. Channel-whitening filters computed. 
Time   7s. Loading raw data and applying filters... 
Time 10.44. Whitened data written to disk... 
Time 10.44. Preprocessing complete!
Time  11s. Optimizing templates ...
Time 19.09, batch 121/132, mu 0.00, neg-err NaN, NTOT 288, n100 3, n200 3, n300 3, n400 3
Time  20s. Running the final template matching pass...
Time 20.44, batch 1/22,  NTOT 0

Index exceeds matrix dimensions.

Error in fullMPMU (line 205)
[~, isort]      = sort(st3(:,1), 'ascend');

Error in testrun (line 20)
rez                = fullMPMU(rez, DATA);% extract final spike times (overlapping extraction)
```

This seems to be a result of mexMPmuFEAT returning empty variables, specifically "st".

At first I thought that this error was a result of spike not being found, but I seems to occur in the optimization stage, not the fitting stage. So I tried to trace it back, to see if the error was really a result of something in fitTemplates.m

I discovered that "dWU" in fitTemplates.m was full of NaNs, which I suspect (but could be wrong) is caused by "uproj" being empty. Is this possible?

I'm using the "fromData" initialization option; in my hands it seems that "uproj" is set full of zeros in line 204 of preprocessData and then emptied on line 287. When I don't use the "fromData" initialization option, I do not get the error above. Any thoughts on why this is happening?

Note, with the "fromData" initialization option, "if isproc(ibatch)" on line 208 of preprocessData is always returning true, so lines 270-275 are not run.

marius10p, 2016-12-21:
Have you tried what I suggested above? Based on your output, it seems that a very small number of spikes are found during optimization (NTOT = 288), which explains why all these templates are NaNs. Try to lower the thresholds to let more spike go through, and also check out DATA to make sure the whitened data did not get messed up in some way.

marius10p, 2016-12-21:
Actually, if this seems to track back to uproj being empty, it seems a different threshold is set too low. Try lowering this one 

ops.spkTh           = -6;      % spike threshold in standard deviations (4)		

maybe to -4? Also try increasing this criterion

ops.crit            = .65;     % upper criterion for discarding spike repeates (0.65)		

for catching more non-repeated spikes with this threshold.

benballintyn2, 2016-12-22:
It turns out that my problem was that I did not have my data in the proper .dat format. As a result the algorithm was reading the incorrect data for each channel and therefore couldn't find any spikes.

closed 2017-03-22 by marius10p

seacritasianman, 2017-05-11:
If you don't mind me asking, I'm having a similar problem. I would just like to confirm that I do i n fact have the data in the proper format. I have the data in a matrix that has dimensions of number channels x number samples. I then use:

fileID = fopen([curr_file '.dat'], 'w');
    fwrite(fileID, curr_dat, 'int16');  %kilosort needs int16 for its data
    fclose(fileID);

to write the data into a .dat file which I then feed into kilosort. This is correct, right?

nsteinme, 2017-05-11:
Try opening the data file you made in neuroscope, which you can download
here: http://neurosuite.sourceforge.net/

You only need to tell it the number of channels in the file (ops.NchanTOT).
If the data looks correct there, then you got it right. If it doesn't, then
try again.