cortex-lab/KiloSort #47: Large amplitude spikes classified incorrectly into multiple clusters

samamckenzie, 2016-12-22:
This issue is best illustrated with a picture (see attached)

Kilosort originally gave me two units (yellow and green in the attached plot) but I noticed that there was unphysiologically high 0ms correlation. I then noticed that the green unit showed two distinct clusters which, when manually split revealed the source of the high correlation (lavender cluster). Notice in the attached cross correlations that the lavender and yellow show many 0ms correlations and share a refractory period: they are the same spikes.  Lavender and green do not share a refractory period: they must originate from different neurons. 

This issue happens every time there is a large amplitude unit (or any large amplitude noise, actually) and is easily spotted by identifying: unusually high 0ms correlations without any jitter, bimodal waveform clusters that, when split, reveal an absence of a common refractory period. This manual stage is, in my opinion, essential for the software to be unusable for our purposes, and easily adds double the manual sorting time as compared to the klustakwik pipeline.  Failing to inspect and correct for this issue, unfortunately, takes the best signal (large amplitude spikes) and distributes it across many units, making it the worst signal for looking at the correlations between single units. 

I understand the root of the problem: variance and mean scale together, so large amplitude spikes are the most variable meaning that the residuals will be highest when looking for template (mis)matches.  Perhaps it is possible to adjust the inclusion threshold according to spike amplitude?

thank you
sam
[double_spike.pdf](https://github.com/cortex-lab/KiloSort/files/669440/double_spike.pdf)

mnpompili, 2018-11-21:
Hi Sam,

we are encountering the same problem here at the Zugaro lab. Many large amplitude spikes assigned to multiple clusters (especially for hippocampal recordings of pyramidal cells). Did you end up finding an explanation/solution for this? 

thanks,
Marco

samamckenzie, 2018-11-21:
Dear Marco

Explanation: Waveform changes over burst.  Mean and variance scale
together. Mean drops with square of distance. So nearby cells will have a
very poor template that will match to waveforms that are smaller than the
first in the burst.  When substracted from the early spikes there will be
energy left over that is then fit to a smaller unit with a waveform on the
same channel(s).  The issue is always between the best signal and the worst

Solution:  You need to look at every pairwise CCG for every unit on the
same shank.   When you see the 0lag no jitter synchrony, you can bet this
is the problem.  Two solutions. 1) recluster these pairs with klustakwik.
2) manually find the odd large spikes on the small unit.  Both require you
to recalculate the PCs in a common space.  I cannot stand phy. since 1) it
isn't naturally organized by shank and 2) the PCs are local and therefore
plotting two units together is meaningless.  Of course it is open source
and both can be fixed, but I am still using klusters to clean up the
units.  we have conversion scripts if you need them

sam

mnpompili, 2018-11-21:
Hi Sam,

we also wrote conversation scripts to keep using klusters and neuroscope. It looks like either us or you could have avoided it...
We have this problem of simultaneous spikes a lot (see below, e.g. clusters 4 and 6 have a peak at 0ms with all other units)
[image]
It looks that the problem is reduced when forcing kilosort to split more its clusters; I'm running some tests now but I also have the problem of not enough memory on the GPU if I asks for too many clusters.

Another option would be to not allow kilosort to detect simultaneous spikes. We could not find an option to disable it but maybe @marius10p would know whether this is possible?

samamckenzie, 2018-11-21:
Yes to more clusters. You can also force the classification closer to the
mean. You don't want to eliminate the overlap, that will defeat the purpose
of kilosort.

Re GPU memory, you have this problem even when you decrease the GPU memory
in the param file? We have never had this problem even with 256 channels.
What model GPU. We can send a param file if you need one.

All the best
Sam

<https://user-images.githubusercontent.com/45228967/48844354-e4405280-ed99-11e8-8142-d27b95a79614.png>

brendonw1, 2018-11-21:
I had an idea in relation to the big spikes problem.
Basically this is due to a desire to catch burst spikes, right?  But of course burst spikes are smaller than the modal spike... the big spikes are of course bigger than the mode.  So one idea is to find the mode, and allow in putative burst spikes smaller than that mode but then use some cutoff to exclude large spikes above that mode/peak of distribution.  Of course you have to allow some spikes bigger than the peak, but there may be some disciplined way of looking at the right side of the distribution and finding a cutoff that for the most part allows in real spikes and excludes huge ones.  This also fits with theory.

By the way I also think we may want to actually go back after all is done (even klusters/phy) and have an algo specifically hunt down burst spikes by specifically looking for reduced-amplitude versions of initial spikes at a few ms delay.

Finally, I modified klusters to allow it to work with >24hr data sets (wasn't displaying time axis right so couldn't look at stability)... if anyone wants that version let me know.

Brendon

petersenpeter, 2018-11-21:
In my experience, I believe the biggest issue is with artifacts. With artifacts, you can have other units "templated" on top of the noise. Between similar sized units I rarely see it. mnpompili, I believe it could be related to your very low threshold for spike detection (ops.Th = [3 6 6]). From my experience having it below [6 10 10] is not necessary/good as you end up with a ton of clusters that cannot be cleaned anyhow. 
These events where Kilosort adds the same spike time stamps to multiple clusters happens at large artifacts, and you should toss the noise artifacts as these are the spurious timestamps contaminating your CCGs. Further, if you experience the problem with similar sized waveforms, I believe it has to do with how you are doing your cluster cutting. Lets say that Kilosort creates two templates that has matching timestamps, meaning that two units' spiking overlaps in time, if you separate the first unit into two separate units, one of these separated units will have matching timestamps with the secondary original cluster. So when cluster cutting, you should often only keep the main cluster, and discard the other spikes as they are already captured with another template.

mnpompili, 2018-11-22:
Hi Peter
We had to lower the threshold because with [5 10 10] and also [4 8 8] a lot of spikes were not extracted. With [3 6 6] we actually do not get too many clusters nor artifact spikes: on a given electrodes bundle (4 to 8 channels) on average we end up with 20 good clusters cleaned out of 40 provided by kilosort. I have to specify  that 1) we have a script that get rid of artifact clusters before we actually open the clu file in klusters so it makes cleaning way easier 2) before running kilosort we detect large artifact events like the animal bumping its head etc.. and we set the signal during these events to zero to avoid these very large event contaminating our clusters. 
 
My problem is rather that I would like to split even more the clusters I obtain because they are contaminated by spikes belonging to other cells. However I cannot do it because my GPU runs out of memory if I ask for more clusters (see above).

Therefore we do not think that these zero lag Xcorr come from artifact spikes. And I do not think it's a problem coming from my treshold. Below you see the same clusters extracted with a treshold at [5 10 10] and the zero lag correlations are still there
[image]

brendonw1, 2018-11-22:
What is the script for cleaning noise spikes ahead of time?  Something you'd want to share?  I think kilosort is great but is a bit lightly supported considering Marius et al are genuinely busy.  So it's be great to come together as a community as much as possible... git repos etc.  (I have some ideas myself but getting recordings going in the lab before focusing on sorting details)

brendonw1, 2018-11-22:
Totally understood of course... thanks!

mnpompili, 2018-11-27:
To come back to the original issue, our current plan is to write a script to get rid of these. This because we are interested in cells that fires together and this clustering behavior will create a lot of perfect co-firing between clusters that would be mostly artifactual. Our script would look for each spike associated to multiple clusters and assign it to the cluster with the most similar average waveform.  It seems fairer to us to decide that a given peak belongs only to one cell.

Is anybody aware of some setting in the Kilosort native scripts to avoid it detecting simultaneous spikes? Such a setting would spare us the time to develop the script described above....

brendonw1, 2018-11-27:
The only think I know of that would minimiize almost-simultaneous spikes is
to reduce the temporal duration of the template... but that would only
handle double spikes more distant in time.

We'd also thought of a script to reassign the double spikes to at least one
of the templates... as I think you're implying you could, if sufficiently
confident also assign a lagged spike to a second cluster.  It'd be great
for the community to have that function.

Maybe we could make a community repository of add-ons to Kilosort?...
probably mostly matlab based since kilosort is matlab-based?