cortex-lab/KiloSort #48: Extract relevant channels of each cluster

georgedimitriadis, 2017-01-04:
Hello Marius,

I am trying to figure out how to retrieve the information that the phy template-gui is using to create the WaveformView with the blue and grey channels. 

I cannot seem to find a way to tell which channels are the relevant ones for each cluster (the ones that phy paints blue). 

I assume since phy is using only the info in the .npy files that kilosort generates that this info is there somewhere.

Could you help me on this?

Thanks

nsteinme, 2017-01-04:
I believe phy calculates the "masks", i.e. what to color blue, based on which channels of the template have an amplitude that is at least some percentage of the max amplitude. @rossant, could you comment?

rossant, 2017-01-05:
We've been trying various ways of choosing the "best channels". Currently I take the channel with the largest template amplitude, and then the N=16 closest channels on the probe. See [the code here](https://github.com/kwikteam/phy-contrib/blob/master/phycontrib/template/model.py#L520-L524).

This is when the templates are saved in a dense format. We will switch to a sparse format later, in that case it will be up to KiloSort to decide what are the best channels during the export.

georgedimitriadis, 2017-01-05:
Thanks for the input guys.

I have been making my own quick and dirty gui to view the channels of the spikes in a cluster, averaged over spikes and because I was not spreading the timeseries over a probe layout the noise in some irrelevant channels was in the way of me seeing what was going on.

Following Marius comment I now allow the user to set a threshold (percentage of the max - min of the template the spike's cluster represents) and then show only the averaged raw data channels that are above this threshold in the template channels. That seems to do the job. It is nice to see that keeping the threshold constant over clusters changes the number of channels visible. It gives an idea of how many channels contribute to the cluster (in relevant terms).
Adding info on place for these channels on the probe is the next step (it will help in deciding whether to keep or throw away the cluster as noise which is the purpose of the gui). 

I was noticing when using the template-gui that sometimes the highlighted channel group was missing channels with obvious spiking activity and sometimes it was adding channels that were obviously noise. Now I understand that behaviour :)

Again thanks for the info

nikolaskaralis, 2017-03-26:
Hi,
could you give an example of how to extract the peak amplitude channel for each cluster?
Essentially I would like to get the same number that is displayed on phy as channel assignment for each cluster.