cortex-lab/KiloSort #49: Double counting?

nicklesica, 2017-01-18:
I think it would be helpful if one could change the snippet length to avoid cases like the one shown below. This happens a lot in IC responses to speech. Also, I'm confused by the spike at zero in the cross-correlogram ...

[image]

nsteinme, 2017-01-18:
Hi Nick, you can set the snippet length by "ops.nt0", which is in units of samples, and by default is 61 samples. I can't guarantee that it won't crash if you try to use something else there, but I think I remember that someone had asked for this before and Marius made it work. 

In general you can get templates like the blue one when two neurons have such high rates (630000 spikes in the red one) that there are lots of instances of the two neurons spiking close in time with a certain offset - so it becomes efficient for kilosort to describe a lot of variance with this sort of combined template. The issue was described before - https://github.com/cortex-lab/KiloSort/issues/10 - and we've talked about possibly having a way for kilosort to identify when one template looks like the sum of two others, but no such thing is implemented yet. There was a good idea for a workaround suggested by @ereming - https://github.com/cortex-lab/KiloSort/issues/41 - but I'm not sure how it went. 

I think the CCG peak is likely to have arisen like this: on some of the instances of the two spikes together (blue), the second neuron (deeper one) was relatively small amplitude but the first one (superficial, same as red) was relatively large amplitude. Kilosort picked one amplitude to best fit, and the result was that there was a residual spike left at the position of the first neuron that was exactly the same shape as the first neuron but half the size. Then this residual got picked up by the actual red template. The prediction of this theory is that the red spikes involved in those close coincidences have particularly low amplitude values. But it is probably not worth the time to really chase it down...

ereming, 2017-01-18:
It seems like manually blanking out the compound spike templates works well, but it would probably be good to test it out on a simulated dataset and compare to ground truth. It's definitely what I would do in this case.

@nsteinme, you're saying the ccg peak at 0 is not directly related to the double spikes right? I have seen these as well, but I haven't remembered to check for them *before* blanking out the compound spikes. In any case, I think you're right that it's not a big deal - I would guess that peak probably represents < 0.1% of spikes.

nsteinme, 2017-01-18:
Well if you had a template with just one neuron in it, then the amplitude that kilosort finds will be pretty accurate and the residual will be quite small - so you should not get the same spike detected twice in that case (i.e. and no CCG peaks). My hypothesis is that in this case the residual ended up big enough to get a double detection _only because_ there was a compound template which forced kilosort to fit a compromise amplitude. I could be wrong though, it's just the only way I could think of to make sense of that CCG peak.