cortex-lab/KiloSort #50: Spike detection in dataset with electrical stimulation

hendrikrth, 2017-01-23:
Hi, I have a dataset recorded with a multichannel electrode with 16 channels. Simultaneous to the recording, electrical stimulation was applied in close proximity to the recording electrode. This electrical stimulation is clearly visible in the raw data on all channels. I wonder if these stimulation artifacts make it even possible to detect spikes in the data. Do you have any opinions on that?

catubc, 2017-02-02:
Hendrik. If I may suggest a simple solution, if the artifact is on all your channels just subtract common means across all channels should mostly remove your artifacts before spike sorting.

marius10p, 2017-02-09:
As @catubc suggests, you could do common average rejection, or just do the artifact detection yourself if you know exactly what this artifact looks like. In principle, Kilosort will remove that through the channel-whitening operation, so just give it a try.

hendrikrth, 2017-04-20:
Sorry, that it took me so long to reply. I computed the mean artifact per channel and removed this from the artifact per channel. After doing so, Kilosort still detected these artifacts as spikes and did not really detect anything else except these artifacts. My next approach was the following. I highpass filtered the data and interpolated the times of the artifacts, I then feed the highpass-filtered and interpolated data into Kilosort. This seems to work, as I now detect the spikes I see in my raw data.

ntotah, 2021-04-28:
Just a follow-up on this since it is still open. We have foot shock artifacts and like the above example, KS2 found only 4 units (and all 4 were artifacts). We used default settings for KS. When we pass the first 30 min of the recording which did not have stimulation, we get 45 units that are all very nice clear single units. The artifacts seem to cause a problem for KS2. Mean subtraction won't work by the way in our case. There is a baseline shift during the stimulation pulse train and that baseline shift differs across trials.