cortex-lab/KiloSort #52: Having trouble installing on Ubuntu 16.04

magland, 2017-02-01:
Could you help walk me through the process. I installed cuda via
sudo apt install nvidia-cuda-dev

I've tried to follow the instructions but when I run mexGPUall I get:

>> mexGPUall
Warning: Version 7.0 of the CUDA toolkit could not be found. If installed, set MW_NVCC_PATH environment variable to location of nvcc
compiler.
 
> In mexGPUall (line 12) 
Error using mex
No supported compiler or SDK was found. For options, visit  http://www.mathworks.com/support/compilers/R2015b/glnxa64.html.

Error in mexGPUall (line 12)
    mex -largeArrayDims mexWtW.cu

magland, 2017-02-01:
Some more information that may be helpful:

>> mex -setup
MEX configured to use 'gcc' for C language compilation.
Warning: The MATLAB C and Fortran API has changed to support MATLAB
	 variables with more than 2^32-1 elements. In the near future
	 you will be required to update your code to utilize the
	 new API. You can find more information about this at:
	 http://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html.

To choose a different language, select one from the following:
 mex -setup C++ 
 mex -setup FORTRAN

marius10p, 2017-02-01:
Hi Jeremy,

Each version of Matlab requires by default a certain version of CUDA toolkit. This can be changed via Matlab's configuration files, but it's easiest to just install that particular version, in this case 7.0.  You can find it here 

https://developer.nvidia.com/cuda-toolkit-archive

Best wishes,
    Marius

magland, 2017-02-11:
Thanks Marius for the reply. It took me a while to get back to this.

Unfortunately it seems that cuda-7-0 and cuda-7-5 are not installable/supported in ubuntu 16.04. I upgraded to matlab 2016b, but again it requires 7-5.

I was stuck until I saw this post by Carl Schoonover: https://groups.google.com/forum/#!topic/phy-users/g0FSHRI0Nao

I am now able to run mexGPUall successfully -- hurray!
Building with 'nvcc'.
...
MEX completed successfully.

But I am nervous about whether it will work.

Do you have a minimal (hello world type) example I can run?

Then my next question would be. Suppose I have, say, a 4x1e7 timeseries array that I wanted to sort. It's in a variable X in matlab. Is there a first pass way to run sorting?

Thanks so much,
Jeremy

marius10p, 2017-02-11:
Thanks for pointing out that post to me, I will link it in the readme file. 

The minimal working example is in the eMouse folder. It creates its own data, sorts it, and benchmarks it. 

The simplest way to use a variable that you already have in Matlab is to just write it to a binary file on the disk. It should be int16, and can be raw data. Kilosort will read it in, process it and set several flags that it needs later. 

You should to follow the eMouse example for building a channel map file for it. For 4 channels this should be straightforward, but you should double check that the format is right. 

Finally, you should change the configuration file (cfg). In particular you need to set the number of clusters (Nfilt).

magland, 2017-02-13:
Thanks for the tips. I'm still struggling with cuda on my particular machine, but that's independent of kilosort, so I'll work that out. Once I get cuda running properly with matlab, and with my particular gpu device, I will revisit.

For now I am trying to run master_eMouse.m with the CPU (useGPU = 0). However, it appears that optimizePeaks.m requires gpu (calls gpuArray). Is there a workaround for this? I am hopeful since it appears this is just during the initialization.

marius10p, 2017-02-13:
Thanks for pointing that out. I will make sure optimizePeaks can run without a GPU.

For now, go to line 49 of config_eMouse.m, 

ops.initialize      = 'fromData';    %'fromData' or 'no'	

and disable the initialization

magland, 2017-02-13:
Thanks. It seemed to work. I also have the same question asked recently by apaks. What are the fields in the rez.mat? ops,xc,yc,xcoords,ycoords,connected,Wrot, etc. In particular I am looking for the timestamps of the events and the corresponding cluster assignments.

Also, maybe this is a question for the phy forum, but I tried to open the results in phy (set it up using miniconda, etc), but encountered the following error upon launching template-gui:

```bash
(phy) magland@jm:~/tmp/eMouse$ phy template-gui params.py
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
WARNING: b'X Error: BadValue (integer parameter out of range for operation) 2\n  Extension:    154 (Uknown extension)\n  Minor opcode: 3 (Unknown request)\n  Resource id:  0x0'
14:23:07 [W] _qt:147              b'X Error: BadValue (integer parameter out of range for operation) 2\n  Extension:    154 (Uknown extension)\n  Minor opcode: 3 (Unknown request)\n  Resource id:  0x0'
14:23:07 [E] cli:39               An error has occurred (RuntimeError): context could not be created
```

marius10p, 2017-02-13:
I will add information for all the fields soon; Kilosort is normally meant to be exported to Phy. 

For your specific questions, timestamps are the first column of rez.st3, while cluster assignments are 2nd column. The waveforms are rez.Wraw, spatial SVDs of these in rez.U and temporal SVDs in rez.W. 

For the Phy question, please ask there, as I don't know exactly what is wrong. @rossant might be able to help.

magland, 2017-02-13:
Thanks. Regarding the ops.initialize = 'fromData' vs 'no'. I see that if the option is set to 'no', the function initialize_waves0.m is called. It appears to initialize things based on a set number of channels (not sure?) so it would it be usable from a non-eMouse dataset? Like from a tetrode?

Just trying to decide whether to wait until you fix the non-gpu code or to modify the initialize_waveforms0. Thanks.

marius10p, 2017-02-13:
The number of channels is determined either from the channel map file, or overwritten by the user via an option. Change the eMouse cfg file to your specifications or make a new channel map file following the example in createChannelMap

nsteinme, 2017-02-14:
Hi Jeremey, use the "rezToPhy" function to produce a bunch of npy files
from the Kilosort results, then you can find their contents carefully
documented here:
https://github.com/kwikteam/phy-contrib/blob/master/docs/template-gui.md

You will also need the npy-matlab repository (user readNPY to load any of
those files into matlab): https://github.com/kwikteam/npy-matlab

rossant, 2017-02-14:
> Also, maybe this is a question for the phy forum, but I tried to open the results in phy (set it up using miniconda, etc), but encountered the following error upon launching template-gui:

Are you running this over SSH? phy doesn't work when you launch it remotely.
Do you have up-to-date graphics drivers?

marius10p, 2017-03-22:
I have now added a lot of information related to each field in the Matlab results file. Please re-open this issue if there are outstanding problems.

closed 2017-03-22 by marius10p