cortex-lab/KiloSort #53: Trouble running master_eMouse, makeGPUall

Sarruedi, 2017-02-06:
I have some issues trying to build the proper NVIDIA CUDA Compiler and wonder whther anyone had the same issue. I tried to run the mexGPUall using CUDA 7.5 in various Maltab version (2015a, 2015b, 2016a) and it only seems to have worked in 2016a since I get the message below when I run mexGPUall. However when I then try to to run master_eMouse I still get an error message about CUDA_ERROR_ILLEGAL_ADDRESS (please see below)

Is the mexGPUall done properly if one get the message below?
And what might be the cause of the illegal address?
I followed all steps described in readme_win_linux.txt and wonder whether there is anything else that has to be done. (Windows7, GeForce GTX 1080, CUDA7.5, Visual Studio Community2013 )

2016a:
>> mexGPUall
Building with 'NVIDIA CUDA Compiler'.
C:/Users/SRL_KiloSort/Documents/KiloSort/CUDA/mexMPmuFEAT.cu(98): warning: variable "NTOT" was declared but never referenced
C:/Users/SRL_KiloSort/Documents/KiloSort/CUDA/mexMPmuFEAT.cu(144): warning: variable "Ci" was declared but never referenced
….
MEX completed successfully.
 
Warning: An unexpected error occurred during CUDA execution. The CUDA error was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Error using gpuArray/filter
An unexpected error occurred during CUDA execution. The CUDA error was:
an illegal memory access was encountered

Error in my_conv2 (line 47)
        S1 = filter(gaus, 1, cat(1, S1, zeros([tmax, dsnew2(2)])));

Error in make_eMouseData (line 71)
    dat = my_conv2(dat, [tsmooth chsmooth], [1 2]);

Error in master_eMouse (line 21)
make_eMouseData(fpath, useGPU);

marius10p, 2017-02-07:
Hi, everything seems normal as far as I can tell. We have used the same combination of GPU/CUDA version/Matlab version. Did you change any of the settings of eMouse by any chance?

I don't think I have ever tested it on Win7, but I don't really see why that would be a problem.

Sarruedi, 2017-02-07:
Hi Marius,
I only changed the paths to correspond to location of the folders of the codes such as KiloSort ,npy , eMouse and the the path for the simulation on my SSD drive.

I got this answers from the Mathworks forum and wonder whether it is about Windows 7:  http://www.mathworks.com/matlabcentral/answers/323661-an-unexpected-error-occurred-during-cuda-execution-the-cuda-error-was-cuda_error_illegal_address?s_tid=email_ans_new_ans_ans_h#answer_253727

nsteinme, 2017-02-07:
> cuda-error-was-cuda_error_illegal_address?s_tid=email_
> ans_new_ans_ans_h#answer_253727

I'm using windows 7 and Matlab 2016a on the computer I use for most
kilosorting. That one has a gtx1080 and driver version 368.81. Is there any
other setting I can report about my machine that might help to clarify?

I wonder whether it could relate to what monitors are plugged into the gpu
- are you using it also to drive your display? I do this, so it should work
in general, but if you're having problems you could try running the display
off the motherboard builtin graphics.

Sarruedi, 2017-02-07:
I tried both running the display on the builtin graphics as well as on the
gpu.
The error still persists.
I guess I will try another graphics card as a next step since I'm really
unsure where the error might emerge at this point.
Worst worst case scenario it's back to run the code on the CPU, which works
but indeed is quite slow in comparison.

Sarruedi, 2017-02-08:
In case anyone has the same issues as I experience, I got the following feedback from the Mathworks forum:
http://www.mathworks.com/matlabcentral/answers/323661-an-unexpected-error-occurred-during-cuda-execution-the-cuda-error-was-cuda_error_illegal_address?s_tid=email_ans_new_ans_ans_h

I don't know why this issue seems no to occur on Nicks machine (lucky you!). 
I will try to reinstall everything using the 368.81 driver or simple switch to Linux.

nsteinme, 2017-02-08:
Sounds like Windows 10 might also work? (and I believe it's what Marius
uses for development, so perhaps the safer choice :) )

Sarruedi, 2017-02-09:
In the meantime I tried another Graphics card GeForce750Ti and the code runs properly now without any other changes on the OS etc. So it indeed seems to be an issue from the specific combo of the GeForce GTX1080 and Matlab2016 on the machine that I am using. I will try the 1080 with the Matlab2017 Matlab release again, which should come out sometimes next month.

marius10p, 2017-02-09:
Have you tried the instructions in 
https://github.com/cortex-lab/KiloSort/blob/master/Docs/readme_win_linux.txt

with respect to setting CUDA_CACHE_MAXSIZE and TdrLevel ?

Sarruedi, 2017-02-09:
Yes I updated both settings as described but the code still throws the error when one tries to use the filter function on the GTX1080. I don't think it's a memory issue since the same settings work for the other card which is officially supported by Matlab whereas the 1080 is not yet.

marius10p, 2017-03-22:
Did the new Matlab release fix this issue?

Sarruedi, 2017-03-22:
I haven't tried yet since we have some issues with the network license for
2017a. The good news is that can run KiloSort with older versions of Matlab
on my actual Ephys data with the GeForce 1080. It's only the demo data that
throws the error. The sorting speed and the quality of the results of your
algorithm is simply amazing! Great fun to work with, especially for big
data sets!

marius10p, 2017-03-22:
Good to hear you have a good experience with Kilosort on your data!

So the error was thrown in the creation of the simulation. Strange, because the main algorithm does use the same "filter" function. I will leave this issue open, and maybe I'll get to debugging that error at some point, if we can replicate it.

Sarruedi, 2017-03-22:
Sure, I will also try it with 2017a once that's possible for me. For now I
enjoy sorting 150Gb at the time and try to understand the different
measures for the quality control.

MHRosenberg, 2017-05-17:
I have encountered what appears to be the same or a very similar issue with the following hardware:
-64 bit windows 10
-Nvidia GeForce GTX 1060
-Matlab 2016a

ALSO, I had the same issue with getting the mexGPUall command to work when I tried it in Matlab 2017a. I suspect that using Matlab 2017a requires using CUDA 8 instead of 7.5. 

I will attempt to run KiloSort on my data and hope that I'll have the same experience as the OP. 

Error:

 master_eMouse
Warning: An unexpected error occurred during CUDA execution. The CUDA error
was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error
was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error
was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error
was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error
was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Warning: An unexpected error occurred during CUDA execution. The CUDA error
was:
CUDA_ERROR_ILLEGAL_ADDRESS 
> In make_eMouseData (line 6)
  In master_eMouse (line 21) 
Error using gpuArray/bsxfun
An error occurred during PTX compilation of <image>.
The information log was:

The error log was:

The CUDA error code was: CUDA_ERROR_ILLEGAL_ADDRESS.

Error in my_conv2 (line 51)
        S1 = bsxfun(@rdivide, S1, cNorm);

Error in make_eMouseData (line 71)
    dat = my_conv2(dat, [tsmooth chsmooth], [1 2]);

Error in master_eMouse (line 21)
make_eMouseData(fpath, useGPU);

MHRosenberg, 2017-05-22:
My error above was resolved by switching to CUDA 8.0 and Matlab 2017a.

danielnee2, 2017-06-16:
Hi Marius:
I also get a similar error while testing with eMouse. (Win 10 64 bit), GTX1060, Matlab 2016b
"
 
Error using gpuArray.randn
An unexpected error occurred during CUDA execution. The CUDA error was:
an illegal memory access was encountered

Error in make_eMouseData (line 62)
        enoise = gpuArray.randn(NT, Nchan, 'single');

And I found the problem to this:

https://www.mathworks.com/matlabcentral/answers/303228-cuda_error_illegal_address

GTX 1060 uses Pascal structure that is not supported by CUDA7.5, and is not solvable until CUDA8.0
 
(which is only supported not until Matlab2017a).

The problem resolved after changing to 17a / CUDA 8.0.
Maybe adding these to the Readme File would be beneficial to the community,
Txs for building such a cool software.

marius10p, 2017-06-16:
Ok, thanks. I updated the readme to the latest Matlab and CUDA versions.

closed 2017-06-16 by marius10p