cortex-lab/KiloSort #54: Fields in rez.mat file

apaks, 2017-02-13:
Hello,

i wanted to ask what each field in rez.mat file corresponds to?

Thanks,

nsteinme, 2017-02-14:
Instead of interpreting them directly, we recommend using the "rezToPhy"
function to save out the npy results files. Their contents are carefully
documented here:
https://github.com/kwikteam/phy-contrib/blob/master/docs/template-gui.md

apaks, 2017-02-15:
I do use phy template gui and know about these files. I just wanted to use rez file in my data analysis pipeline, it is easier to extract everything from one file and there is additional info in rez that is not visible in phy eg raw waveforms. 
I think it would be useful for users to know what these field mean:
[u'U',
 u'W',
 u'Wraw',
 u'Wrot',
 u'WrotInv',
 u'cProj',
 u'cProjPC',
 u'dWU',
 u'errall',
 u'fW',
 u'iMega',
 u'iMegaC',
 u'iNeigh',
 u'iNeighPC',
 u'mu',

Thanks,

marius10p, 2017-03-22:
I have added information about most of these fields in the main readme file. The ones I ignored are not consequential, but only temporary fields that I might need for analyzing results of the algorithm, trying out improvements etc.

closed 2017-03-22 by marius10p