cortex-lab/KiloSort #55: Post-processsing "matrix dimensions" error

catubc, 2017-02-15:
Hi

Thanks for this package, it's working great so far.  I've been running it on shorter recordings without problem.

I ran it on a 100GB+ recording and it exited with some matrix error. However, as far as I can see the .npy spiketimes and unit IDs are correct, perhaps the .mat file is messed up?

No rush on this, just wanted to report it.

Thanks!
catubc

********************************************************************

`>> master_eMouse

pathToYourConfigFile =

/media/cat/12TB/in_vivo/tim/cat/2017_01_31_barrel_ephys_ophys/sort_alltrack/kilosort

Time   0s. Loading raw data... 
Time 699s. Channel-whitening filters computed. 
Time 699s. Loading raw data and applying filters... 
Time 2113.56. Whitened data written to disk... 
Time 2113.56. Preprocessing complete!
Time 2128s. Optimizing templates ...
Time 3781.23, batch 37661/37680, mu 25.14, neg-err 1575953.918397, NTOT 2312151, n100 8575, n200 2277, n300 0, n400 0
Time 3784s. Running the final template matching pass...
Time 6854.59, batch 6201/6280,  NTOT 3860480
Error using  < 
Matrix dimensions must agree.

Error in merging_score (line 23)
    sum(hs2(1:m2)<mmax/troughToPeakRatio);

Error in merge_posthoc2 (line 89)
            mo = merging_score(f1old - f2old, f1new-f2new, ops.fracse);

Error in master_eMouse (line 57)
rez = merge_posthoc2(rez);
 
java.lang.IllegalArgumentException: Comparison method violates its general contract!
	at java.util.TimSort.mergeHi(Unknown Source)
	at java.util.TimSort.mergeAt(Unknown Source)
	at java.util.TimSort.mergeForceCollapse(Unknown Source)
	at java.util.TimSort.sort(Unknown Source)
	at java.util.TimSort.sort(Unknown Source)
	at java.util.Arrays.sort(Unknown Source)
	at sun.awt.datatransfer.DataTransferer.setToSortedDataFlavorArray(Unknown Source)
	at sun.awt.datatransfer.DataTransferer.getFlavorsForFormatsAsArray(Unknown Source)
	at sun.awt.dnd.SunDropTargetContextPeer.getTransferDataFlavors(Unknown Source)
	at sun.awt.datatransfer.TransferableProxy.getTransferDataFlavors(Unknown Source)
	at java.awt.dnd.DropTargetContext$TransferableProxy.getTransferDataFlavors(Unknown Source)
	at com.mathworks.widgets.grouptable.transfer.TransferUtils.unwrapObjectList(TransferUtils.java:84)
	at com.mathworks.widgets.grouptable.transfer.TransferController.isDragToSelectTransferable(TransferController.java:383)
	at com.mathworks.widgets.grouptable.transfer.TransferController.access$200(TransferController.java:48)
	at com.mathworks.widgets.grouptable.transfer.TransferController$DropListener.dragOver(TransferController.java:236)
	at com.mathworks.widgets.grouptable.transfer.TransferController$DropListener.dragEnter(TransferController.java:227)
	at java.awt.dnd.DropTarget.dragEnter(Unknown Source)
	at sun.awt.dnd.SunDropTargetContextPeer.processEnterMessage(Unknown Source)
	at sun.awt.X11.XDropTargetContextPeer.processEnterMessage(Unknown Source)
	at sun.awt.dnd.SunDropTargetContextPeer$EventDispatcher.dispatchEnterEvent(Unknown Source)
	at sun.awt.dnd.SunDropTargetContextPeer$EventDispatcher.dispatchEvent(Unknown Source)
	at sun.awt.dnd.SunDropTargetEvent.dispatch(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.trackMouseEnterExit(Unknown Source)
	at java.awt.LightweightDispatcher.processDropTargetEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$200(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
>> `

marius10p, 2017-02-17:
Hi, 

Looks like it failed in the final merging step. You should still be able to export the results to Phy without that. The java errors make it hard to know what went wrong. Perhaps you could try just increasing the Java allocated memory? https://uk.mathworks.com/help/matlab/matlab_external/java-heap-memory-preferences.html If you still have the results in RAM, you can just rerun the final step, no need to run the whole thing again.

~Marius

catubc, 2017-02-18:
Thanks Marius, I've upped that sucker, it was set at 380MB for some reason. I'll see how it runs next time.

Thanks!
Catalin

closed 2017-03-20 by marius10p