cortex-lab/KiloSort #56: Spike2 format (.smrx) or .mat to .dat or .bin

arnal46, 2017-02-27:
Hello,
I'm starting to use your software and I really like how it works. 
The problem is my data is acquired using CED Spike2 its stored in .smrx, but inside of their software it allow me to export it in .mat format but not in .dat or .bin. I don't know how your software read the .dat or .bin files. 
Please, if you can explain me how to create the .bin (or .bin) data file from a .mat file, I will really appreciate it.

Thank you :)

marius10p, 2017-02-27:
Hi there, 

The binary files contain the raw data, as an array of int16 ordered like this
tp1 ch1, tp1 ch2, ....  , tp1 chN,  tp2 ch1, tp2 ch2 , ... ,tp2 chN, ....
where tp1 is timepoint 1, ch1 is channel 1, etc. 

You can use the matlab function fwrite to create this file. It is described here 
https://www.mathworks.com/help/matlab/ref/fwrite.html

arnal46, 2017-03-02:
Thank you, it was really helpful!

closed 2017-03-20 by marius10p

NaomiHan, 2017-04-13:
Hi Marius,

I would just like to check if I understand this correctly, so I'm sure I'm giving the script the right data format. Do I need to create an int16 array of (#channels * #timepoints) rows by 2 columns? So in the hypothetical case of 32 channels and 10 timepoints, I write a 320x2 int16 array in a .dat file?

Many thanks

marius10p, 2017-04-13:
Hi Naomi, 

Not sure why you have two columns. In Matlab, it should be an int16 array of  (#channels x  #timepoints) that you write to a binary file. Because Matlab is "column major" the order in which this array will be written is going through all channels for timepoint 1 first, then all channels for timepoint 2, then all channels for timepoint 3, and so on.  This is the order that Kilosort assumes when reading the binary file. Cheers, - Marius

NaomiHan, 2017-04-14:
That clears it up, thanks!