cortex-lab/KiloSort #59: details about standardConfig parameter setting

jerryhanson, 2017-03-27:
In the standardConfig code file, there are some parameter without detailed footnote. I want to know how every parameter will affect the sorting results. Could you help me with this? Is there some document I could gather information from, thanks.
 Especially for this part:

% the following options can improve/deteriorate results. 		
% when multiple values are provided for an option, the first two are beginning and ending anneal values, 		
% the third is the value used in the final pass. 		
ops.Th               = [6 12 12];    % threshold for detecting spikes on template-filtered data ([6 12 12])		
ops.lam              = [10 30 30];   % large means amplitudes are forced around the mean ([10 30 30])		
ops.nannealpasses    = 4;            % should be less than nfullpasses (4)		
ops.momentum         = 1./[20 1000];  % start with high momentum and anneal (1./[20 1000])		
ops.shuffle_clusters = 1;            % allow merges and splits during optimization (1)		
ops.mergeT           = .1;           % upper threshold for merging (.1)		
ops.splitT           = .1;           % lower threshold for splitting (.1)		
		
% options for initializing spikes from data		
ops.initialize      = 'no'; %'fromData' or 'no'		
ops.spkTh           = 4;      % spike threshold in standard deviations (4)		
ops.loc_range       = [3  1];  % ranges to detect peaks; plus/minus in time and channel ([3 1])		
ops.long_range      = [30  6]; % ranges to detect isolated peaks ([30 6])		
ops.maskMaxChannels = 5;       % how many channels to mask up/down ([5])		
ops.crit            = .65;     % upper criterion for discarding spike repeates (0.65)		
ops.nFiltMax        = 10000;   % maximum "unique" spikes to consider (10000)

marius10p, 2017-03-27:
Hi, 

Most of the options in there won't really affect the optimization, unless they are set to some extreme, unreasonable values. 

The options that do matter are Th and lam. To understand exactly how these work, you would need to read over the math in the NIPS paper.  Intuitively, you have the comments in the file: threshold is a threshold, the higher it is, the fewer spikes are detected. lam si a penalty, large penalties force the extracted amplitudes around the mean for each cluster (and low penalties allow for large variability in the mean). 

The options for initializing spikes from data don't matter much. These are just use to initialize Kilosort in a reasonable manner. In some cases this initialization helps,  but in many cases it does not matter at all, because the Kilosort optimization works fine with the annealing provided by the momentum term (ops.momentum).

cindyhfls, 2018-02-01:
Hi,

I am a bit confused what the values of ops.Th and ops.lam are. Do I have a range of number (e.g. 1-100) to choose from, or can I just arbitrarily input a number?

Why do they come in groups of three?