cortex-lab/KiloSort #60: Best standard configs for 16ch data?

michaelaloft, 2017-03-29:
I'm using a 32 (2 x16) channel probe with parallel sites (~25um apart). The two shanks are 250um apart so I want to sort them separately. Can anyone recommend the best parameters for the standard config file? In particular I'm curious about ops.Nfilt - in the version of the code that I have there's a comment next to it which says:  
% number of clusters to use (2-4 times more than Nchan, should be a multiple of 32). 
Is the fact that I've only got 16 channels going to be a problem here?

Are there any other parameters I should particularly careful about given my relatively low number of recording sites?

marius10p, 2017-04-04:
Hi, I think everything should work fine. Usually the default parameters work fine regardless of the number of channels. Other things are more important, like SNR or channel density, but within a reasonable range the default parameters should work.  

 I would recommend processing both shanks together anyway, particularly if the raw data file contains all channels. This way noise clusters can be shared between probes, and it is also easier to deal with the results in the Phy GUI, since you only have one file with all the clusters to load and manipulate. To do this, make sure you make your own customized channel map file, and give the shanks different labels ("kcoords"). That ensures that no clusters will be shared between them.

For 32 channels, try 128 clusters first. If it looks overclustered, try 64. For too many clusters, Kilosort will keep finding smaller spikes, unclusterable etc. but it will also start oversplitting clusters, particularly if a recording has drift.

LauriTK, 2017-12-27:
> To do this, make sure you make your own customized channel map file, and give the shanks different labels ("kcoords"). That ensures that no clusters will be shared between them.

There is a one thing that I don't understand with the kcoords. We use 4 x 8 probes and I have grouped each shank (shanks are 200µm apart), but I still get spikes from neighbouring shanks (seen in Phys WaveformView) for most of the clusters. Is this how it should work, or should kcoords totally isolate shanks from each other? Not sure if I understand all correctly.

clausagerskov, 2018-08-13:
can we get a response? how does kcoords exactly work?

marius10p, 2018-08-14:
That's just a display effect in Phy, where it has to plot blue at least N nearest channels. You can change that parameter N in the nearest channel PCs in the configuration, but it's purely a visual change, and has no effect on the actual sorting.

clausagerskov, 2018-08-14:
kcoords is just a visual effect? so it does not make sure that kilosort treats tetrode groups separately as stated in the createChanMap? 
or is the fact that channels are coloured together a visual effect? shouldnt this be labeled a bug?

nsteinme, 2018-08-14:
No, the other way around: kcoords is real, and in your screenshot it successfully restricted the sorting to channels 24-31. But channels 16-23 got colored blue in phy anyway, since phy will color 16 channels blue by default. The blue is the visual effect but kcoords is real and should have worked - if you have other evidence that the sorting itself didn't work, please let us know (try viewing the template with 'w' and see whether it is not flat on channels from the other shanks). You could also post an issue on phy's github asking for easier control of the blue coloration if you like. Closing this issue for now but feel free to re-open if we didn't get it.

closed 2018-08-14 by nsteinme

Zoe0793, 2019-05-13:
Hi, I have set up Kilosort 2 with a 4*8 silicon probe. How can I change the default in Phy from 16 to 8 coloured channels? I understand its just a visual effect but it is sometimes making it confusing to understand what spike belongs to which cluster in trace view. Thanks! Zoe

hana777, 2020-03-06:
Hi Zoe, not sure whether this is still relevant for you, but it is possible to change the default by installing this plugin:
https://phy.readthedocs.io/en/latest/plugins/#changing-the-number-of-spikes-in-views
The value of controller.model.n_closest_channels corresponds to the number of highlighted channels (in TraceView, ProbeView and WaveformView).

Zoe0793, 2020-03-06:
Hello, I solved the issue already, but thank-you for providing an
alternative solution!