cortex-lab/KiloSort #61: Mac OS 10.11.6, Xcode7, mexcuda

micheleacox, 2017-04-01:
This is an FYI more than an issue. @marius10p feel free to close / file wherever appropriate.  

Our System: We run Matlab 2016a on a MacPro OS 10.11.6 with a NVIDIA GTX 750 TI. Our CUDA driver is version 7.5.27. Our GPU driver is 10.11.14. My understanding is that Matlab does not play nice with CUDA driver version 8, so we have not upgraded. 

The Problem: Some update in the last few weeks--unclear if it was a system update or a particular software involved but did somehow involve Xcode--broke our ability to compile the GUP mex files. This was causing issues for a new user on the system who wanted to operate out of their own "fresh" KiloSort dir, i.e. my previous compiled mex files continued to work fine, but she could not run mexGPUAll in the new dir.   

Anyhow, this is how we fixed it: 

1.      Checked that 2016a is compatible with XCode7
2.      Installed XCode7
3.      Configured mex to use XCode7 à xcode-select –switch /Applications/XCode7.app/
4.      Checked that nvcc is 7.5 (no cudnn)
5.      Updated clang and clang++ xml files in mexopts
6.      Matlab: Ran mex –setup c++
7.      Verified: mexcuda mexGPUExample.cu runs correctly
8.     Cloned KiloSort to new local directory, added to Matlab path, Ran mexGPUAll

marius10p, 2017-04-04:
Thanks! I'll see where in the documentation this should be added.

nsteinme, 2017-07-10:
Thanks very much for this, @MartinKahn ! I was trying to install on a new computer and was only able to get this working after reading your recommendation about which compiler version to use. So let me add a couple updates:

- These instructions also work on Windows 10
- It also works (at least on Win10) if you use visual studio community 2015 WITH update 3, which I tried since that was the only one I found. 
- To get this version of visual studio, what I did was: Go to myvisualstudio (https://my.visualstudio.com/), login with microsoft account, click Downloads and search for "visual studio community 2015". If it tells you there are no products, you may have to enable "visual studio dev essentials" for your account, which is free. 

@marius10p maybe we should consider including precompiled versions of the code for common OS/Matlab/CUDA versions... it really is a huge headache to get the compiler all setup and working... I'm sure there are potential users who get lost at this stage (I have talked to some of them).