cortex-lab/KiloSort #62: Isolated peaks and its settings

leoniecazemier, 2017-04-04:
Hi, thanks for your software! I have a question regarding the isolated_peaks script. When I try to run my raw data through the sorting, isolated_peaks continuously leaves me with an empty 'peaks' array. This is because my sum_peaks array does not have any values under 1.2. In fact, the values in sum_peaks vary between 8 and 43. 

Now, I'm guessing this might have something to do with the loc_range and long_range values that isolated_peaks takes in, but I don't quite understand in what way those variable would change my sorting outcome. I don't just want the software to run, I also want the result to make sense.

So my question basically is: what can I do to not end up with an empty 'peaks'  array and what would be the most sensible way to do so?

marius10p, 2017-04-04:
Hi, let's see what I can do. The isolated_peaks script is only an initialization. If you disable the initialization, that step won't even be performed and in many cases it makes only a small difference. 

You seem to be describing a case of very low SNR data, where there are no clear big spikes, is this true? If not, can you please add here a picture of a raw data trace with spikes? The best thing to do for low SNR data is actually to skip the initialization step altogether.

loc_range and long_range are just parameters for defining peaks: it has to be the max (or min) for plus/minus loc_range samples, and it also has to be the only maximum above threshold in a plus/minus 
long_range  interval. Changes to these parameters should not make a big difference however, particularly if it does not find any peaks at all. 

You can try lowering the threshold if you really need to have the initialization.

leoniecazemier, 2017-04-04:
Hi, thanks for the quick response!

You are quite right about the low SNR ratio; we made a recording where something went amiss and where we seem to get the same MUA on all if not most of the channels. I was running the software to see if we could nevertheless find one or more single units in this weird broad signal. 

So I will try to follow your advice and skip the initialization on my next run and see if that gives better results.

closed 2017-04-06 by leoniecazemier