cortex-lab/KiloSort #63: Duplicate output files: spike_clusters.npy and spike_templates.npy

mspacek, 2017-04-07:
I've noticed that when ks exports results to .npy files, it generates two identical files that contain the cluster IDs of all the spikes, with similar names: `spike_clusters.npy` and `spike_templates.npy`. Is this intentional, maybe for some kind of backward compatibility, or an oversight? I'm wondering which should be considered the correct file name to use in the future. I'm currently reading the `spike_clusters.npy` file when importing ks results into [spyke](http://spyke.github.io/).

rossant, 2017-04-07:
`spike_templates.npy` is static whereas phy modifies `spike_clusters.npy` during manual clustering. Initially, the two vectors are identical.

marius10p, 2017-04-07:
The two vectors may also not be identical initially, if the "posthoc" merging step has been run. In which case Phy will treat the automatic merges as if they were made by a human in the GUI.

closed 2017-04-07 by mspacek