cortex-lab/KiloSort #65: Parameters for tetrode recordings

lshaheen, 2017-05-10:
Hi all,

I have some recordings made using Thomas tetrodes that I'd like to sort using Kilosort. The default parameters have worked pretty well, but often there is a lot of manual merging necessary due to cluster splitting. This happens because I'm recording in the inferior colliculus where there the neural density is high, so there is a lot of multiunit activity but usually only one isolatable single-unit. It would be best to avoid so much merging both because it takes time, and because it's prone to error. In addition when Kilosort over-clusters, some of the clusters that should be merged also contain some spikes that by my eye do not belong (looking at the PCA or template space scatter plots with Phy). An example is attached.

Here are the parameters I've been using:

job.Nchan=4;
job.Nfilt= 32;  
job.nNeighPC=4;
job.nNeigh=4;
job.whiteningRange=4;
job.spkTh=-4;
job.GPU=1;

I think the best thing may be to reduce Nfilt, but setting it to less than 32 causes an error. Any advice would be awesome!

Thanks,
Luke

[image]
[image]

tbrouns, 2017-06-14:
I'm not sure if this reply will be of any use to @lshaheen, considering it's been over a month, but it might be useful to anyone else who is interested in using Kilosort for tetrode recordings. 

In the config file example it says for the "Nfilt" field: 

`% number of clusters to use (2-4 times more than Nchan, should be a multiple of 32)
`

This suggests that we can only use an Nfilt that is greater than 32. where Nfilt = n * 32 (with n being a whole number).  However, I've noticed that factors of 32 also work. So you can use 16 or 8 as well. 

If you choose to use 8 clusters, then you need to set the `showfigures` field to 0, because this will result in an error otherwise.

lshaheen, 2017-06-14:
Thanks for the tip, I'll try that out.

closed 2018-09-10 by marius10p