cortex-lab/KiloSort #66: MATLAB Error: Index exceeds matrix dimensions

seacritasianman, 2017-05-10:
Hello,
I'm trying to sort 64-channel data over 8 shanks. I have the data split shank by shank, and I'm running kilosort on the shanks one at a time (so 8 channels at a time). I've put the kilosort code into a loop to reduce the amount of work I have to do. When it loops into the second iteration (second shank), I keep getting an error that "index exceeds matrix dimensions", and kilosort crashes at the fitTemplates step. Furthermore, in the command window, I get some weird things, like a mu of 0 and neg-err: NaN. Any clue what's causing this? Thanks!

seacritasianman, 2017-05-10:
I would also like to add that it seems to kind of randomly affect certain recordings, but not others. I ran it on a 900s 64 channel recording at 20kS/s with no problems, but on a different recording that was 540s, 64 channel at 20kS/s, it failed. It might be something earlier in the processing pipeline (reading from the Intan .rhd files into MATLAB), but I wanted to see if maybe you've experienced this problem in the past? I write the data into 8 separate files, each with 8 channels of data as 'int16' .dat binary files.

nsteinme, 2017-05-10:
If you can give a line number for the crash, it would be helpful?

Here are two things to try: 1) Try running the second shank just by itself
(skip the first shank) - in that case does it work? It may not be a problem
with the looping but just with the configuration for the second dataset. 2)
If it's a problem with the looping, try putting the kilosort master file
into a function rather than a script, and call that function in a loop.
This will ensure that the workspace it uses is clean, and doesn't have
anything left over from the previous iteration.

seacritasianman, 2017-05-10:
I've tried running the second shank by itself, and it still fails (as well as further shanks). I haven't tried completely shutting down MATLAB, then running the second shank, but I'm worried that's not the issue, simply because the loop works for some data sets but not others (no discernible trend). I'll attach a screenshot of the command window when it crashes (sorry I forgot to do it before haha)
[image]

nsteinme, 2017-05-10:
It might for instance happen when the second shank has more spikes than the
first, or vice versa - sorry but please do try restarting matlab and
running the second shank alone, and/or putting kilosort into a function
within your loop.

seacritasianman, 2017-05-10:
I've tried running kilosort as a function in a loop, and restarting matlab, and it still isn't working. I ran some older data (which works), and one further thing I've noticed is that on the command line during "Optimizing Templates" the values after n100, n200, n300 etc. change numbers, whereas for the dataset that isn't working they all remain 0, does that mean anything?

I've also looked at the raw data as well, it doesn't seem like there should be anything wrong with it, each of the channels seems normal.

seacritasianman, 2017-05-10:
One thing I've noticed with the new data (which doesn't work) as compared with an older data set (which does work) is that the values after n100, n200, n300 etc are different. For the older data, the numbers seem to fluctuate as the batch updates. However, for the new data, it fluctuates a little up until ~ batch 100, then it's just 0 after n100, n200, etc. Does this mean anything?

seacritasianman, 2017-05-10:
Update: I was able to remedy the problem by changing the ops.initialize value from 'fromData' to 'no', as per https://github.com/cortex-lab/KiloSort/issues/44. What are the practical implications of doing this? Thanks.

closed 2017-06-29 by seacritasianman