cortex-lab/KiloSort #69: No Knowledge of Electrode Locations

marsalad, 2017-06-17:
I am ready to send tons of data into KiloSort, but do not know the spatial organization of the electrodes I am simulating. I have run my data through successfully by placing them in a line just to test that everything works, but am now looking for a way to run the data through without location information. I have already tried the following fixes unsuccessfully:

- placing all electrodes at the same xcoords, ycoords, and kcoords (matlab displays warning messages upon completion of KiloSort processes, and Phy crashes upon loading due to a linear algebra error)
- placing all electrodes at the same xcoords and ycoords, but different kcoords (same as above)
- omitting any entry to xcoords and ycoords (KiloSort crashes because a part of the program requires these values)
- omitting a channel map file entirely (this causes no issues, but looking at the code, it looks like the default is to put the electrodes in a line, which is what I was trying to avoid in the first place)

Is there a possible fix for this and I'm just missing something or is location information crucial to running KiloSort?

nikolaskaralis, 2017-06-17:
If I understand correctly what you want to do, it is enough if you do the
following:
xcoords = zeros(nchannels,1);
ycoords = 50 * [1:nchannels];
kcoords = ones(nchannels,1);

This should let KiloSort run (it doesn't use the exact location) and they
will appear in Phy as consecutive channels.

Hope this helps.

Best regards,
Nikolas

nsteinme, 2017-06-18:
Marius can correct me if I'm wrong, but the one place this information can
be used is in the whitening step, where it whitens over the nearest 32
channels by default. But if you set ops.whiteningRange=Inf then I think it
will be as Nikolas says - geometry will not be used.

Nick

marius10p, 2017-06-18:
Like Nick said, xcoords and ycoords will mostly not matter for processing, just for visualization in Phy. However, kcoords does have a big effect, as it restricts neurons to live only on channels with the same kcoords. 

Just to clarify though, is it the case that:
1) multiple wires never record the same cell (like a Utah array), 
or 2) they often do, it's just not possible to know a priori which wires are close to each other.

nikolaskaralis, 2017-06-18:
Hi Marius,
for wire arrays, which would be case #2 above, do you suggest using the
same kcoord for all electrodes?

In the case of tetrodes, is there a considerable gain from having each 4
electrodes with their own kcoord as opposed to putting all channels
together with one kcoord?

auditorycohenlab, 2017-06-18:
Hi Marius

The issue is that the wires CAN record the same wire but we don't know a priori if they do and we don't know the specific spatial relationships between the wires.

auditorycohenlab, 2017-06-19:
sorry to be a impatient but any thoughts out there? thanks! yale

nsteinme, 2017-06-20:
I think your first solution - placing them all on an imaginary line -
should work fine. I think you said that ran successfully, so were you
unsatisfied with the results? If so you might have a different problem, but
in any case more description of what you saw with that test would help, I
think.

marius10p, 2017-06-20:
Yes, try setting kcoords to the same value for all channels, and Kilosort will determine the spatial relationships itself. For tetrodes, yes, definitely same kcoords for all neurons in a tetrode (and different between tetrodes). This is the only way to exploit a unit's distribution over channels.

marsalad, 2017-06-20:
Thanks for the help! It sounds like putting the electrodes in a line will work for us.

closed 2017-06-20 by marsalad