cortex-lab/KiloSort #71: Error line 205 in fullMPMU.m, linked to problem in cpuMPmuFEAT.m

madeleinea, 2017-06-28:
Hi,

I keep getting the error: index exceeds matrix dimensions at line 205 of fullMPMU.m ([~, isort]      = sort(st3(:,1), 'ascend');). After looking into it, I realized that st3 is empty at that point, and that is probably the cause of the error. st3 is defined in lines 188-193, but the program keeps skipping over the if statement between 155 and 202, so st3 remains empty. I think that the real source of the problem is in the cpuMPmuFEAT function, where st (which is the source of st3) is supposed to be created, but it keeps returning as an empty array. The cpuMPmuFEAT.m function seems to be a lot of complicated mathematical computation, so I'm having trouble figuring out why it keeps returning an empty st variable. I'd really appreciate help from anyone who knows anything about this function.

Thanks!

marius10p, 2017-06-29:
Hi! 

Can you please tell me if the array DATA is empty when you get that error? It should be in the workspace. 

It's unusually rare for no spike at all to be found in a batch, which is what your error implies. Hence why I think something might have gone wrong before that. If DATA does exist, and you can see spikes in it, then try lowering the detection threshold (ops.Th = [4 8 8], for example).

madeleinea, 2017-06-29:
Hi,

Thanks for your quick reply. DATA isn't empty when I get that error (I've
included a screenshot of attempting to open DATA at a breakpoint right
before the error (line 203)). I can't see what's inside, I guess it could
be all zeros. What do you think? If it was a problem with my raw data
conversion, how would I know?

I may try lowering the detection threshold in the meantime.

Thanks,
Madeleine

[image: Inline image 1]

madeleinea, 2017-06-29:
Update: lowering the detection threshold did not help, I got the same error
at line 205 of fullMPMU.m.

madeleinea, 2017-06-29:
Actually, I spoke too soon. Lowering the threshold did seem to resolve the
error. Thanks so much!