cortex-lab/KiloSort #74: Error Line in fitTemplates - matrix dimensions mismatch

Sarruedi, 2017-07-12:
I just came across matrix dimension error with one of my data sets and wonder whether the issue could be the caused by a technical issue during the data acquisition with SpikeGLX. During the data acquisition SpikeGLX reported an error as it ran out of space to store the data towards the end of a Neuropixel recording session and I had to kill the application to get it to close. Nevertheless I got a file of 140GB and the pre-processing function did not throw an error, which lead me to believe that the saved data was still fine. 

When trying yo fit the templates I get an error in line 192 of fitTemplates:  
Inner matrix dimensions must agree: 
data =dataRAW * U(:,:)

In the fitTemplate function the iterator i reaches a value of 3866 (while Nbatch*ops.nfullpasses+1 = 43675).
for all i <3866 the size of dataRAW is 25644 x 384 and the size(U(:,:) is 384 2880. So that's fine.
but for i=3866 dataRAW is 25644 x 84, size(U(:,:) is 384 2880, which leads to the error.

Does anybody have an idea why the matrix dimensions suddenly might not match or have some experience with KiloSort and sorting SpikeGLX files that crashed during acquisition?

I expected an error for the last i but not one that is way less than Nbatch*ops.nfullpasses+1.
Any ideas on whether one can save the file?

nsteinme, 2017-07-13:
Could you check whether the number of bytes in the file is a multiple of
2*nChanTot? I think you can do it like this:
>> d = dir('myFilename.imec.ap.bin');
>> mod(d.bytes, 2*385)

If spikeglx crashed in the middle of writing a sample, and only wrote the
last sample for a subset of channels, then that might be what went wrong.
In that case you can make a new file that just excludes the last sample so
that it comes out to a correctly sized matrix.

nsteinme, 2017-07-26:
@Sarruedi did this get resolved? I will close the issue but re-open if not.

closed 2017-07-26 by nsteinme