cortex-lab/KiloSort #75: data file type (neuralynx)

mjk1028, 2017-07-23:
Hello,

i wonder if kilosort takes neuralynx continuously sampled files (*.ncs; 32 or 64 channels). if not, do you have any suggestion what to do to make it work on kilosort? thank you so much for sharing your thoughts in advance!

best regards,
MJ

nsteinme, 2017-07-23:
It looks like Neuralynx provides a matlab importer on their website: http://neuralynx.com/research_software/file_converters_and_utilities/

If you have the data in matlab as a matrix, size (nChannels x nTimePoints), then you can write it to a new file just like this:
```
>> fid = fopen('myNewFile.dat', 'w');
>> fwrite(fid, myDataMatrix, 'int16');
>> fclose(fid);
```
Then give that file to kilosort.

You want to be careful about the datatype: if your data is not originally int16 then it may be easiest to typecast it first; I'm not sure whether kilosort will work with int32 or some other format. But you can try it.

mjk1028, 2017-07-23:
Thank you so much for your quick answer!
I will try it out as you suggested. Have a good one!

closed 2017-07-26 by nsteinme