cortex-lab/KiloSort #78: PCspikes not found

zeeMonkeez, 2017-08-03:
Trying to get started with kilo sort, using the dataset published here: http://data.cortexlab.net/singlePhase3/data/ (using `rawDataSample.bin` for now). Sorting seems to work until I reach this point:

```
Time 752s. Running the final template matching pass...
Error using load
Unable to read file 'PCspikes'. No such file or directory.

Error in fullMPMU (line 98)
    load PCspikes

Error in run_kilo (line 21)
rez                = fullMPMU(rez, DATA);% extract final spike times (overlapping extraction)
```

Searching the whole repository does not find a place where this file is being generated.

marius10p, 2017-08-03:
That file is in KiloSort/configFiles/. Please add the full repository  to the path using addpath(genpath( yourPathToKilosort))

zeeMonkeez, 2017-08-03:
Cheers. Somehow I assumed that would be dataset specific (as a comment regarding `PCspikes2.mat` suggests: "you might want to recompute this from your own data"

closed 2017-08-03 by marius10p