cortex-lab/KiloSort #80: Noisy clusters

shihaisun, 2017-08-06:
Hi KiloSort team,

A lot of my clusters are noisy following KiloSort spike-sorting. Below is an example of a noisy cluster (as seen in the feature view and waveform view). 

[image]

I changed the maximum clusters in the config file to 128 and left the rest on default. Can you guys recommend me any setting changes to fix my noisy clustering issue?

Thank you so much for your time,
Scott

nsteinme, 2017-08-07:
The big vertical lines on the waveform view suggest that you have some very
large-amplitude artifacts. You might try detecting these with a simple
threshold and then removing that part of the data, then run again. If you
don't think that the artifacts impacted your sorting results much, though,
then you can simply "split" them out in your phy gui. Looks to me like that
would be fine for the example cluster you're showing.

brendonw1, 2018-03-09:
Hey @nsteinme any nice ways you guys have to run such a filter?  Maybe a plugin?

nsteinme, 2018-03-10:
No, I don't have a method for this - I always run common average
referencing before sorting, which I guess must be removing any such
artifacts that I have. I rarely see problems like this.

brendonw1, 2018-03-10:
Ah i see.  That's a good point.  So you create a second binary file that's averaged across channels and then run kilosort on that?  (also, have you ever tried median filtering?).  Thanks

nsteinme, 2018-03-10:
Yes, I use this function:
https://github.com/cortex-lab/spikes/blob/master/preprocessing/applyCARtoDat.m

No, I haven't tried median filtering...

brendonw1, 2018-03-10:
Thanks!
By the way it looks like that does 2D median filtering actually! 
 "% Subtracts median of each channel, then subtracts median of each time
% point."
(and all calls are "median" not "mean")

nsteinme, 2018-03-10:
I think by "median filtering" people sometimes mean like what matlab's "medfilt1" does - where it takes a running window and replaces each point with the median of the values in that window. Here I'm doing something different: First, for each channel separately, I subtract the channel's median (across all time) from every point in the trace (this step just removes channel offsets, which is a thing with neuropixels probes but probably irrelevant for most probes); second, for each time point separately, I take the median across channels and subtract it from every channel's point. The second step is what people call common average referencing (whether it's a mean or median, probably doesn't matter too much...). I should have made that comment in the function more clear!

Alkohost, 2018-03-21:
Hello!
I just started using Kilosort and faced with noisy clusters also. I mean all detected clusters had no gaps around zero at autocorrelograms. I used [Klusters, NeuroScope, NDManager] - tools before with KlustaKwik for a few channels (usually 4), and there same data had more clear clusters. Can you please tell me what parameters should I change for cluster clearing?

g874259, 2018-10-26:
Hi @nsteinme,
1. Should we do the data preprocessing (subtract channel's median and CAR) ONLY for good channels?
2. The data preprocessing on 1. should be done after high-pass (300Hz) filtering?
Tks, JJ