cortex-lab/KiloSort #81: templates span separated channels

adede1988, 2017-08-07:
I'm using the eMouse simulation script. I changed the coordinates of my simulated electrodes to be 1.5mm apart and I set my kcoords variable to be a different number for every channel. Despite this, I'm getting templates that are non-zero across multiple channels.

marius10p, 2017-08-08:
Sorry, that simulation is not meant to look at channel separation, since we use fixed templates obtained from a real experiment. The only things you can do with the simulation are: 
1) test that your installation of Kilosort is working at all, 
2) see how noise amplitude, number of neurons, firing rates affect performance. 
You would need to replace the base templates with single channel ones (if you can find the place inside the simulation script where I load the base templates from disk).

adede1988, 2017-08-08:
Ok, sounds good. I think I'll go ahead and just try it out on some data!

closed 2017-08-08 by adede1988