cortex-lab/KiloSort #82: st3 variable is empty

adede1988, 2017-08-09:
I'm trying to run kilosort on 6 channels of data collected on single wires. Three of these channels are extremely noisy. One has a single very high rate cell. One has 2 cells, and one has at least one cell, but perhaps more. 

I input kcoords values such that each channel is its own group. The data are in a binary file. 
Ops.nNeighPc=1
ops.nNeigh=1
ops.whiteningRange=6
ops.maskMaxChannels = 1

all other values are default. 

I get an error when I run fullMPMU. Specifically, st3 is empty and I get an index exceed matrix dimensions for: 
[~, isort]      = sort(st3(:,1), 'ascend');

I tried using an if statement to skip the line if st3 is empty, but the problem propagates down the code. What is variable st3 supposed to hold? What does it mean that it's empty? Thanks for your help on this!

adede1988, 2017-08-09:
Oh, I just saw that another user had this same issue. My DATA variable is not empty, so I'll try lowering the detection threshold and see what happens. Thanks!

adede1988, 2017-08-09:
It seems that kiloSort cannot handle the noise channels. When I removed the noise channels and ran only on 3 good channels, it was able to run without throwing an error. However, it cannot open in phy. This seems like a separate issue though

erosco, 2017-11-03:
I'm having the same issue: my set-up became unplugged for a minute or so during the recording so I have a period of very high-amplitude noise.  I've adjusted the spike threshold, but depending on the value it's set to, I end up with either an empty st3 variable or clusters made up only of high-amplitude noise and no real spikes.  Any suggestions for how I can limit spike detection to only the lower-amplitude spikes, or manually cut out the offending time period?

nsteinme, 2017-11-03:
To manually cut, in matlab:

```
% read in the data
fid = fopen(filename, 'r');
dat = fread(fid, [nChansTotal Inf], '*int16'); % only if you can fit all of your data into memory. Otherwise, you have to do it in chunks, or use memory-mapping
fclose(fid);

% set offending time period to zeros
dat(:,noiseStartSample:noiseEndSample) = 0;

% write a new file with the cleaned data
fidOut = fopen(outputFilename, 'w');
fwrite(fidOut, dat, 'int16');
fclose(fidOut);

```

hth -

Bionerdess, 2020-02-10:
dat = fread(fid, [nChansTotal Inf], '*int16'); % only if you can fit all of your data into memory. Otherwise, you have to do it in chunks, or use memory-mapping

How do you do it in chunks or use memory mapping? I'm running out of memory during this step using kilosort (not manually cutting).

Error using fread
Out of memory. Type HELP MEMORY for your options.

Error in RawBin_to_PostKilo_HEADER_NEWTTL_TRIM (line 58)
     hs = fread(fid,[nchans,inf],'int16=>int16');
 
58       hs = fread(fid,[nchans,inf],'int16=>int16');