cortex-lab/KiloSort #83: Assertion Error

adede1988, 2017-08-09:
kiloSort seems to run fine, but I get an "Assertion Error" when I try to open the phy gui. I can open the phy gui when I open it on simulated data from the eMouse, but it won't open when I'm trying to run it on my data. Any thoughts on things I might try? I've already rerun kiloSort from scratch a second time and tried again, but to no avail.

adede1988, 2017-08-09:
update:   File "c:\users\adam\miniconda3\envs\phy\lib\site-packages\phycontrib\template\gui.py", line 353, in _init_data
    assert all_features.ndim == 3
AssertionError

adede1988, 2017-08-09:
update: the error originates from pc_features.npy having 2 as opposed to 3 dimensions. The code is poorly commented though, and it's hard going to trace these errors without knowing what these variables are supposed to contain or where they're generated.

adede1988, 2017-08-27:
I am still unable to move past this error.

nsteinme, 2017-08-27:
Did you change anything about the PC features in the config file? This line:
dd                  = load('PCspikes2.mat'); 
If you did, I suspect you used something that's the wrong size? 

Another possibility is that zero spikes have been found - what is the size of spike_times.npy? 

Yet another possibility is that you only have one channel in your data and there's a bug related to that - how many channels in your data file? 

pc_features is supposed to be size [nSpikes, nFeaturesPerChannel, nPCFeatures] where nFeaturesPerChannel=3 and nPCFeatures is whatever you set ops.nNeighPCs to be. Given that information can you identify which dimension is the missing one?