cortex-lab/KiloSort #85: Clusters with high similarity - Peak Amp offset in time

Sarruedi, 2017-08-18:
Sometimes I get clusters which have a high similarity score (eg. 0.9) but the waveforms seem shifted in time in the waveform view as the waveforms are not peak aligned in the display in phy. Now I am unsure whether I should still merge these clusters based on the waveform similarity/amplitude/pc features or whether the offset in time is actually indicative of being a different unit. The data has been sorted in KiloSort and it might be caused in the sorting step in KiloSort or simply be a display issue in phy.

Based on the ACG/PC features for such clusters, I think the clusters actually arise from same unit but it seems that they either end up with a different timestamp for the peak in LioSort or display in phy does not align spikes based on the peak amplitude but some earlier bin. 

Does anyone know how the waveforms are put in relation to each other in phy? 
How could highly similar clusters not be peak aligned in time in relation to each other?

greggheller, 2017-09-15:
I have had this problem as well. 

I think it is because of the way kilosort updates the templates. If there is a predictable voltage change before the spike, then that gets incorporated into the template. The templates/waveforms must be aligned in phy by the beginning of the template rather than the peak time. 

It is nice that the similarity score still identifies these clusters as similar but it would be really nice if we could align the waveforms/templates by the peak time. This would make the decision much easier when other indicators are ambiguous.  

I am also interested in learning how the similarity score is calculated, and would like to second Sarah's questions.

greggheller, 2017-09-15:
Here are two examples
[image]
[image]

LongLabGit, 2017-10-02:
We have encountered the same issue: 
[image]
[image]

nsteinme, 2017-10-03:
Hi all - sorry for the very slow response on this. 

First a clarification: phy (the GUI software) does not do anything with aligning waveforms, it simply displays the waveforms according to the timestamps it is given. So the fact that you get two templates with similar shapes but mis-aligned is on kilosort's end. 

But spike alignment is a really hard problem - there's plenty of discussion in the literature about this. Kilosort takes the typical approach of aligning to the negative peak, which runs into problems when there are two channels that have similar peak amplitudes but are offset in time - that appears to be the case with channels 51 and 63 in the LongLab example. The red cluster are the spikes where 63 (the earlier site) were larger amplitude, and the blue cluster is those for which channel 51 were larger amplitude. 

There's really nothing to do about this except curate it manually - so yes, do merge the red and blue in @LongLabGit example. @greggheller yours is a bit harder to evaluate since we're only seeing the final merged result, but I guess you get the point. 

With spikedetekt2 we used to take a different approach to alignment, using a center of mass. This solved some situations like this, but had its own set of weird edge cases, so I think in the end we weren't particularly motivated to keep it for kilosort. 

So I'm going to close this issue, but if you have any particular ideas/suggestions for solving this, do feel free to open a new issue. 

To answer one other specific question in this thread, the similarity score is the cross product of the two template waveform shapes. It would come out even higher if the waveforms were properly aligned, but comes out all right just because the waveforms are still largely overlapping, even when shifted.

closed 2017-10-03 by nsteinme

greggheller, 2017-10-06:
Thanks for your reply Nick. I can definitely see why it would be much harder than I had first imagined, thank you for your patience and taking the time to explain.

The only thing I can think of would be to align by the earliest/latest minimum out of the channels with similar amplitude to the peak channel. But I think this would have its own set of problems and the misalignment occurs so infrequently that it isn't worth worrying about at all.