cortex-lab/KiloSort #86: Strange Error in Whitening Step

madeleinea, 2017-09-14:
Hi, I'm getting an odd error in the svd function of the whiteningLocal file (see attached picture). It's odd because I've sorted hundreds of files with Kilosort and I'm suddenly getting this issue with this one file. I can't tell what this would be about, input to SVD should be the same with any file. Please let me know if you have any information on this error, because this could be an important file for me. 

Thank you!

[image]

herbat, 2017-10-19:
Hi, I'm getting the same error. Just reporting in so this gets more attention!

basimons, 2021-02-08:
Just to comment for other people having this issue, but you probably have Nrank > Nchannels, so if you reduce your Nrank it probably fixes it.