cortex-lab/KiloSort #87: Issues with post-hoc merges

jsiegle, 2017-09-27:
The `merge_posthoc2` function (commented out by default) works great in general, but I'm running into two problems.

1. The similarity scores don't match after post-hoc merging. It seems like `rezToPhy` doesn't take into account the new cluster IDs.

2. Increasing the number of templates (896 vs. 512 initially) results in an error:

```
Error using  < 
Matrix dimensions must agree.

Error in merging_score (line 23)
    sum(hs2(1:m2)<mmax/troughToPeakRatio);

Error in merge_posthoc2 (line 89)
            mo = merging_score(f1old - f2old, f1new-f2new, ops.fracse);

Error in kilosort_master_file (line 28)
rez = merge_posthoc2(rez);
```

The same thing happens with 1024 templates. Any idea what could be causing this?

nsteinme, 2017-10-03:
Hi Josh - @marius10p will have to address your second question, but for the first one I believe this is a phy issue. The similarity score is always between templates, and in phy when you merge two clusters it will update similarities as the max of any constituent cluster similarities - i.e. if you merge A and B, then the similarity of this new cluster with C is max( sim(A,C), sim(B,C) ). But I'm not sure whether phy is successful at making this correct new similarity score if you didn't do the merging within phy itself. Try posting an issue about it on phy's issue tracker.

jsiegle, 2017-10-03:
We actually solved the second issue by adding the following after line 17 of `merging_score.m`:

```
if size(mmax,1) == 0
    if size(hs1,1) > 0
        mmax = max(hs1);
    elseif size(hs2,1) > 0
        mmax = max(hs2);
    end
end
```

The errors were occurring when either hs1 or hs2 was an empty array. Everything seems to work after adding this check, although I'm not 100% sure there are no unintended side-effects.

I just submitted an issue to the phy repo: https://github.com/kwikteam/phy-contrib/issues/132