cortex-lab/KiloSort #89: Possible for one spike to be in multiple clusters?

shihaisun, 2017-10-04:
Hi KiloSort team,

Just wondering if it's possible for KiloSort to assign one spike into multiple clusters? For example below, could these three clusters share the same spikes? I'm asking because of what the cross correlograms show.

[image]

Thanks,
Scott

nsteinme, 2017-10-04:
It is possible for kilosort to identify spikes from multiple clusters as
occurring at the same time sample, i.e. kilosort definitely can decide that
cluster 14 and cluster 55 both had a spike at time=10.3432 sec. This will
happen when kilosort identifies a spike of one of them at that time point,
subtracts it away, and finds that the remaining signal still looks like the
spike of the other cluster. This allows it to detect when multiple spikes
really do happen at the same moment in time. I would say that seems likely
to be what's happening here, as the red and blue clearly have different
waveform shapes (red much narrower). In other words, I don't think this is
an error of any kind, I think it is correct behavior. Whether you are
confident enough to interpret the lag=0 CCG bin as meaningful will be up to
you - if it is critical, I suggest you carefully examine the raw data
around these coincident spikes as much as possible and see whether you
believe kilosort has done a good job.

shihaisun, 2017-10-04:
Thank you so much for the prompt response! Closing the issue.

closed 2017-10-04 by shihaisun