cortex-lab/KiloSort #91: Importing spike data from ntt files

herbat, 2017-10-20:
Hello KiloSort team,

I'm trying to import ntt files to KiloSort, which isn't directly supported, but I can extract spike times and waveforms from .ntt. I tried to pass the data simply as a dat file, but it's clearly not the right way to go. 

Can you offer some guidance as to how I should insert the spikes into the kilosort process?
Or, it might be of more use to others as well, can you specify what you expect from the binary(.dat) files to contain?

Thank you!

nsteinme, 2017-10-21:
The binary file should contain a 2D matrix representing the voltage over
time on each channel. If you have that 2D matrix (nChannels by
nTimeSamples), you can write it to a file like this:

fidOut = fopen(outputFilename, 'w');
fwrite(fidOut, dat, 'int16');
fclose(fidOut);

hth -

herbat, 2017-10-21:
Thank you! I suppose the data should be continuous, so that's why it doesn't really work with triggered tetrode files.

nsteinme, 2017-10-21:
You could try simply concatenating all the waveform snippets, or try
creating an empty matrix of the correct size and adding the snippets to it,
i.e. reconstructing the raw data given only the parts you recorded. If you
try it, let us know how it goes.

herbat, 2017-12-12:
I ended up writing an embedding function, which is under testing. If there is interest, I'll commit it here when I'm sure it works.

closed 2017-12-12 by herbat

julienfiorilli, 2018-04-18:
Hi herbat,

>  I ended up writing a converter function, which is under testing. If there is interest, I'll commit it here when I'm sure it works properly.

Curious to here your experience/progress with this, I am in the same situation:  spike times and waveforms from .ntt

Did you manage to convert them into .dat file? Did it work for you?

Cheers!

brykko, 2018-04-19:
I'd be interested to know whether this works. I suspect that some parts of the algorithm (e.g. whitening) might behave pretty differently when dealing with zero-padded spike snippets instead of continuous raw signals, but maybe that's not so important, I'm not sure.

But for anyone doing recordings with Neuralynx acquisition hardware, I would strongly recommend opting for continuous raw recordings instead of using the .ntt files. Kilosort's template-matching algorithm is really effective at extracting low-amplitude spikes which would not have been detected by the simple amplitude-based triggering used for .ntt files. Also, I often see misaligned (and therefore truncated) waveforms in .ntt files, which will be detrimental to the spike sorting.

If anyone is interested, I have a dat converter for the Neuralynx raw .nrd format: https://github.com/brykko/nrd2dat.

sandraromerop, 2019-08-07:
> @julienfiorilli
> Hi,
> I did successfully convert some ntt data to dat, but it seemed that KiloSort doesn't behave the way we expected. It only recognised a little subset of the spikes, and the clustering didn't work either.
> However, it worked fine for continuous recordings(neuralynx and openephys), and it was already planned for the lab to make the switch, so we don't record ntt anymore.

hi Herbat,

Can you tell me how you conver the .ncs files to the apropriate data format to input it to Kilosort? The neuralynx codes load it into matlab as ADC , how do you convert it to nCHannels x timeStamps  int16 format ?