cortex-lab/KiloSort #93: Exclude Electrodes

RoscoeOBrian, 2017-10-26:
Hi,
I have a 6x10 multielectrode recording (MCS). Unfortunately I have large noise on one of the electrodes. Is it possible to exclude one of the electrodes when it comes to detecting and sorting the spikes?
I could also think of feeding only 59 electrodes in the analysis, but then there is a "gap" in the channelmap. Can Kilosort handle this?

thx

nsteinme, 2017-10-26:
Just exclude the entry in the channel map that specifies the dead site's
row in the data file. It should work fine. Note that while Nchan is now
going to be 59, NChanTOT still needs to be 60 (assuming no other data is in
the raw file) so that the file can be loaded properly. Give it a try and
let us know if it looks like a problem?

RoscoeOBrian, 2017-10-27:
ok, I tried to exclude electrode 56. And it kind of works:

`%  create a channel map file

Nchannels = 59; %original: (60)
connected = true(Nchannels, 1);
chanMap   = [1:55 57:60]; %original: (60)
chanMap0ind = chanMap - 1;
%xcoords   = ones(Nchannels,1); %Original code
% removed Electrode 56: 7 
xcoords   = [5	5	5	4	4	4	3	3	3	2	1	2	1	2	1	1	2	1	2	1	2	3	3	3	4	4	4	5	5	5	...
    6	6	6	7	7	7	8	8	8	9	10	9	10	9	10	10	9	10	9	10	9	8	8	8	7		7	6	6	6];

%ycoords   = [1:Nchannels]'; %Original code
% removed Electrode 56: 2
ycoords    = [3	1	2	1	2	3	1	2	3	1	1	2	2	3	3	4	4	5	5	6	6	4	5	6	4	5	6	5	6	4	...
    4	6	5	6	5	4	6	5	4	6	6	5	5	4	4	3	3	2	2	1	1	3	2	1	3		1	2	1	3];
`
In the template-gui, I also see a gap, where 56 is supposed to be, BUT it seems like electrode 57 is labeled 56 now, 58 is 57 and so on till 59 (instead of 60).
So, although I defined the channel map `chanMap   = [1:55 57:60]`, kilosort creates a `[1:59]` namelist somewhere ... 

In the StandardConfig-file I omitted "ops.NchanTOT" and "ops.Nchan".

So, I can rename my electrodes posthoc, but it is getting really compolicated, when I want to exclude several channels...

nsteinme, 2017-10-27:
Sorry, my fault - the easier approach is just to set:
`connected(56) = false;
`
Leave chanMap and x/ycoords the same. The way to think about it is, chanMap/connected/xcoords/ycoords all have the same number of elements, one per physical site. x/ycoords gives the position of the site. chanMap specifies the row of the data file to find data from the site. connected specifies whether to include it or not (maybe we could have called this "useChans" or something).

RoscoeOBrian, 2017-10-27:
ok,
I get the following error:
`Error using  + 
Matrix dimensions must agree.

Error in preprocessData (line 154)
            CC        = CC + (datr' * datr)/NT;

Error in master_file_example_MOVEME (line 19)
[rez, DATA, uproj] = preprocessData(ops); % preprocess data and extract spikes
for initialization`

It seems, that the variable "CC" (GPU related) is still 60, but "datr" and "dataRAW" are corrected for the noisy channel and therefore 59.

nsteinme, 2017-10-27:
Hm, on line 23 ops.Nchan should have become 59, so that when CC was
initialized on line 98, it was 59x59. Can you determine where it has gotten
the wrong value of ops.Nchan? You need ops.Nchan=59 and ops.NchanTOT=60.

RoscoeOBrian, 2017-10-27:
ok, it runs again. But the Problem with the electrode names remains: 55 should be gone, 56 to 59 should remain.

[image]

nsteinme, 2017-10-27:
This might now be a confusion about 0-indexing? Phy uses zero-indexing, but
Kilosort/matlab does not. So connected(57) refers to the channel that phy
will call 56. If that doesn't explain it, can you clarify which channel you
set to be disconnected, and which one phy is not showing you?

RoscoeOBrian, 2017-10-27:
As far as I see it, the calculations are correct.
I want to exclude e56 (phy calls it 55). Exactly this electrode (x7, y2) is omit in the results. The Noise is also gone - everything fine so far. 
But for some reason Phy continues using the electrode name 55 (for e56 here) and omits 60 instead. 

Perhap `chanMap=[1:55 57:60]` in the createchannelmapfile would solve it?

nsteinme, 2017-10-27:
I see what you're saying. I never noticed it, but it does that for me too in phy - just numbers the channels in order 1 to N, rather than skipping the numbers of the ones that are excluded. 

You can try playing around with the channel map maybe - I would do it not by re-running kilosort but by just writing a new channel_map.npy. Then delete the .phy folder and re-open phy. If you can't get it to work then I think this is a bug, and you should make a new issue over at phy's page - specifically at phy-contrib: 
https://github.com/kwikteam/phy-contrib/issues

Since this seems to be a phy issue now, I'll close this issue, but please re-open if I've got it wrong. Sorry to pass the buck, but it'll have to be @rossant who fixes this phy gui issue if you need it fixed.

closed 2017-10-27 by nsteinme