cortex-lab/KiloSort #96: parameter setting for linear array with electrode distance of 50 microns

WaveformDai, 2017-11-17:
Hi, 

I'm now having a linear array electrode on which the distance between two adjacent electrodes is 50 microns. It's desired that combining information from multiple nearby electrodes may help extract a single unit. I tried setting ops.kcoords from one group to several different groups, and I also change the number of ops.Nfilt, but there still remains too much manual splitting and merging to do.

I found similar issue reported in https://github.com/cortex-lab/KiloSort/issues/65, but it still didn't give any suggestions on how to set proper parameters and what parameters are really crucial on such condition.

Below figure is selected from a recent biorXiv paper of Cadena et al. It describes how they separate their electrodes into different virtual electrode groups and the distance between their electrodes is 60 microns.

[image]

Any suggestion on how to set parameters like the one described above or in other ways that can help sort single units is really appreciated!

nsteinme, 2017-11-17:
Hi, we'd recommend keeping all electrodes in one group (ops.kcoords = ones(nChannels,1)). The electrode grouping is intended for situations when it's impossible to see one neuron on other channels, as when you have multiple shanks or tetrodes. Could you clarify whether you are having too much merging, or splitting, or both? If you have a great deal of electrode drift then you may just have a very difficult/impossible problem - otherwise we recommend setting ops.Nfilt so that there is merging but no splitting, since merging is faster and more accurate than splitting.

WaveformDai, 2017-11-27:
Most of further manual clustering we needed to do was splitting rather than merging. Below figures come from one typical example( ops.Nfilt is set to twice the number of electrode). It's obvious that current pc features can be separated into two different clusters and one cluster always lies on the x/y axis or on the origin in most pc projection subplots(Figure 1-3). 

Figure 1
[image]

Figure 2 
[image]

Figure 3
[image]

Apart from some easy cases, I also encounter other cases in which I'm not sure if I should do more manual splitting or merging and I hope you can give me some advice.

1) KiloSort assigned some spikes to one cluster(Figure 4) but their PC feature can obviously be separated into two clusters( with different waveforms and no clear refractory period in CCG. Figure 5). However, their template feature seems to form a continuous distribution( I think that's the reason why KiloSort will assign them to one cluster. Figure 5).  Should I split this cluster into two different clusters?

Figure 4
[image]

Figure 5
[image]

2) KiloSort assigned some spikes to two different clusters. Although they have distinct waveforms and their CCG doesn't show explicit refractory period, they belong to the same electrode and their pc features have large overlapping. Besides, their template features also seem to form a continuous distribution(Figure 6). Should I merge these two clusters?

Figure 6
[image]

After uploading these figures, I'm aware this may take you too much time to read. So I really thank you for your generous help.

nsteinme, 2017-11-28:
>    1. KiloSort assigned some spikes to two different clusters. Although

It looks to me like things have worked reasonably well in the examples
you're showing. In your figures 1-5, the amplitude of the spike is really
quite close to the noise level, and the clusters that seem obvious to you
in PC space are quite blurred together really. I think it is not surprising
that Kilosort did not find much point in splitting small, close-to-noise
points apart.

In your figure 6, I would recommend not to merge based on the lack of
refractory period in the CCG and the fact that blue is larger amplitude
than red on all channels (not consistent with electrode drift, in which red
would be larger than blue above, and blue larger than red below). If there
are more than just those three channels then it will help. The shape is
also not matching enough for me to think it might be bursting adaptation,
though I don't know what species/brain region you are recording from, so
who knows. I agree this case is not clear-cut.

hth -

WaveformDai, 2017-11-28:
Thanks for your detailed reply and I'm really grateful for your help. As for the inconsistency of amplitude of red/blue clusters between AmplitudeView and WaveformView, I remembered there is one note in the online document of TemplateGui (see attached figure) and I think that's the reason why there will exist such an inconsistency.

[image]

xf123chen, 2017-12-07:
Hi,WaveformDai,
I come from china and use the multi-channel recording systems. I try to setup this software, but didn't success. can you help me,thanks. xf123chen@163.com

cindyhfls, 2018-01-19:
Hello, we were using electrodes with inter-contact distances = 50 microns as well, but we were having the problem that most of our LFP's were not removed, and the waveforms did not look like neural activity but resembled noise.

[image]

nsteinme, 2018-01-20:
I agree that the data you show here looks only like noise. If you load the
data yourself separately and it looks like something besides noise, then
you may have the incorrect number of channels configured in ops.NChanTOT
and params.py n_channels_dat. But instead I think here it looks like this
is just not neural data to begin with.

Note that phy TraceView will display the contents of the original file,
whereas kilosort does whitening and high-pass filtering first before
analysis. So what you see in TraceView does not match exactly what kilosort
worked with.

cindyhfls, 2018-01-23:
Hi Nick! 

Thanks for your timely reply!
I have just loaded the original .ns5 file from the Ripple recording System. Perhaps I should figure out some way of only loading the channels I wanted into a separate binary file?

Btw, have there been any successful attempts in Kilosort with contacts with distances > 200 microns, and in non-human primates?

I am not so sure whether it would be worth trying that out in our dataset. Since it seems to work best for small contacts and n-trodes etc.

Best,
Cindy

nsteinme, 2018-01-23:
I have used kilosort on NHP data with 150µm spacing of contacts. At that
distance, spikes on a given contact have little or no amplitude on adjacent
contacts. Still, there's not much to lose by using kilosort anyway: it will
still run faster and be more accurate than other things I've tried, plus
resolving overlapping spikes. It would be the algorithm of choice in my
opinion for that dataset - but you could certainly try multiple things and
compare. If you do, let us know what you think!

cindyhfls, 2018-01-23:
That's a good news to hear! Do you mind if we contact you if we encounter further problems?

xf123chen, 2018-01-23:
*{
        margin: 0;
        padding: 0;
    }
    font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
Hi,Kilosort is absoutly very good to spike sort. I use Kilosort for tetrode electrode(Neuronexus), site distance of one tetrode eletrode is about 10um. Before,I use offlinesorter software to sort, when the tetrode electrode have rarely unit and S/N is good. It's ok. but when the tetrode electrode have many electrode, . It's nearly don't  auto-sort. I take many times to manual sort according to waveform shape in offlinesorter.  I often despair in sorting data.Until, I learn kilosort. It's computer very fast in tetrode data. I check result of kilosort. It's very same with my manual sort result in offlinesorter.Thanks kilosort, Thanks cortex-lab, Thanks everyone for kilosort. 
about site distance,I think distance>50um don't tetrode. I suggest you look paper(György Buzsáki, Large-scale recording of neuronal ensembles).
cxf

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    
    
        font{
            line-height: 1.5;
        }

xf123chen, 2018-01-23:
*{
        margin: 0;
        padding: 0;
    }
    font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
sorry,I write error.but when the tetrode electrode have many electrode---wrongbut when the tetrode electrode have many Units---right

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    
    
        font{
            line-height: 1.6;
        }
    

    
        font{
            line-height: 1.6;
        }
    

    
        font{
            line-height: 1.5;
        }
    
-----------------------------------------chen xfLaboratory of Learning and Memory, Key Laboratory of Animal Models and Human Disease Mechanisms, Kunming Institute of Zoology, CAS, Kunming 650223, China ----------------------------------------

                

    On 1/23/2018 23:44，xf123chen<xf123chen@gmail.com> wrote： 

    *{
        margin: 0;
        padding: 0;
    }
    font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
Hi,Kilosort is absoutly very good to spike sort. I use Kilosort for tetrode electrode(Neuronexus), site distance of one tetrode eletrode is about 10um. Before,I use offlinesorter software to sort, when the tetrode electrode have rarely unit and S/N is good. It's ok. but when the tetrode electrode have many electrode, . It's nearly don't  auto-sort. I take many times to manual sort according to waveform shape in offlinesorter.  I often despair in sorting data.Until, I learn kilosort. It's computer very fast in tetrode data. I check result of kilosort. It's very same with my manual sort result in offlinesorter.Thanks kilosort, Thanks cortex-lab, Thanks everyone for kilosort. 
about site distance,I think distance>50um don't tetrode. I suggest you look paper(György Buzsáki, Large-scale recording of neuronal ensembles).
cxf

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    
    
        font{
            line-height: 1.5;
        }

nsteinme, 2018-01-24:
@cindyhfls yes, please feel free to post other issues here if you run into any.

@xf123chen thanks! glad you have had success with Kilosort.

closed 2018-01-24 by nsteinme