cortex-lab/KiloSort #97: Can I use Kilosort for these datasets?

ferchaure, 2017-11-22:
Hi, I have a few questions:

- Can I use Kilosort for single channel or tetrodes datasets?
- Is it possible to use detected spikes as an input?

marius10p, 2017-11-22:
Yes.

No, but some people write the clipped spikes into a binary file in series, which Kilosort can take in. It's really not ideal, since you cannot resolve waveform overlaps  very well this way.

ferchaure, 2017-11-22:
I tried to sort an eMouse simulation of one channel, but this error occurs during the sorting:

> Time 22.98. Preprocessing complete!
> Index exceeds matrix dimensions.
> 
> Error in optimizePeaks (line 119)
>     W(:,j,:) = w(:, 1:Nrank)/sum(Sv(1:ops.Nrank).^2).^.5;
> 
> Error in fitTemplates (line 32)
>         WUinit = optimizePeaks(ops,uproj);%does a scaled kmeans
> 
> Error in master_eMouse (line 25)
> rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively

ferchaure, 2017-12-01:
Just before that error, in optimizePeaks.m is the line:
https://github.com/cortex-lab/KiloSort/blob/d748f1e45f42d53fbf86dbefb39e8dd8ce08ca26/initialize/optimizePeaks.m#L125

For one channel ops.Nrank should be 1, doesn't it?  But that line set it equal to 3

ferchaure, 2017-12-01:
And probably the 3 inside the reshape is related to Nrank to
https://github.com/cortex-lab/KiloSort/blob/d748f1e45f42d53fbf86dbefb39e8dd8ce08ca26/mainLoop/mexMPregMUcpu.m#L12

kkatlowitz, 2017-12-01:
In the meantime, as a workaround, I just pad my data with 2 extra channels that only have low white noise. As expected, Kilosort only finds spikes in my single channel