cortex-lab/KiloSort #102: discard artifacts

marius10p, 2017-11-30:
For example as brought up in #98. This could either be a pre-processing step on the raw data, or a step during template matching to ensure spikes are not matched if they have too much variance orthogonal to the template. 

Any other ideas welcome.

czuba, 2018-12-06:
What should our expectation be for any further development of kilosort?

It seems like many fundamental aspects/issues (such as this one) have been left totally unfinished. Not meaning to take a jab at the creators (...people get busy, projects/funding move on, etc), just trying to consider whether its wise for us to build internal infrastructure around it.

marius10p, 2018-12-06:
Artifact removal is one of the most modular parts of the pipeline, so I encourage anyone who wants to write that code and contribute it. 

However, I encourage users to build their own artifact removal preprocessing, because the artifacts can be very different from lab to lab. There is no all-encompassing artifact removal step that will fix everyone's problems, but perhaps there could be multiple variants that users could choose from.