cortex-lab/KiloSort #104: crash due to RAM limit?

czawora, 2017-12-04:
Hi,

I am running Kilosort on a Mac with Matlab2017a. I am trying to sort 1 hour of a 96 channel recording, file size ~20GB. Very early on after loading the data, Matlab + a kernel process start using a ton of memory (~120GB) and Matlab crashes. I am wondering if this just a RAM limitation ( I have 64GB of RAM and specified that in the config ) or something that would be fixed by using a GPU (right now I do not have a CUDA-compatible GPU). Thank you for your help.

I am attaching the printout before it crashes. 

Best,
Chris

[image]

czawora, 2017-12-04:
The crash still occurs on a 10 min recording with ops.parfor = 0 and ops.initialize = 'no'. Are there other steps where I can reduce the memory usage?

marius10p, 2017-12-04:
Hi, this is strange. The only major use of RAM is in storing the data, so you should be fine. You can control that via 

ops.ForceMaxRAMforDat   = 20e9; % maximum RAM the algorithm will try to use; on Windows it will autodetect.

Try to set to 0, but that's probably not it. I don't know where the memory leak could be, since we only use built in Matlab functions for the CPU code.  Sounds like you are in Linux? How many filters are you using (ops.Nfilt) ? Does your error happen in the initialization phase? Looks that way, so try ops.initialize      = 'no' (that's the default).  Also try to stop it at line 101 in fitTemplates.m and run it line by line to see what operation overflows the RAM.

czawora, 2017-12-04:
One thing that did prevent it from crashing, but did make the processing super slow, was setting 

ops.NT                  = 128*256+ ops.ntbuff	

Thanks for the advice though! I will see how your recommendations affect memory

czawora, 2017-12-04:
I am running on OS X

czawora, 2017-12-04:
When running with

ops.initialize = 'no' 
ops.NT = 128*1024+ ops.ntbuff

the script crashes at line 199 in fitTemplates.m 

 [dWU, ~, id, x,Cost, nsp] = ...
            mexMPregMUcpu(Params,dataRAW,fW,data,UtU,mu, lam .* (20./mu).^2, dWU, nu, ops);

marius10p, 2018-09-10:
I did not see this, but it looks you are asking for an extremely large number of clusters (~10,000). Since you have 96 channels, you should really only ask for a few hundred clusters.

closed 2018-09-10 by marius10p