cortex-lab/KiloSort #106: New Kind of crash issue

danielnee2, 2017-12-09:
Hi All:
I read through all the discussion threads and 
did not see the same error I got here.

[image]

We are using a 256 channel (Multichannel systems) MEA for recording,
and I convert the data into .dat binary files for kilosort input.
At shorter recordings kilosort works find,
but when the .dat file is exceeds 90 Gbs at Nfilt = 256,
it crashes at line 271 during the preprocess.

I am using a Win10 system with 32 Gb ram. Anyone suffers from similar issues?
Thanks

marius10p, 2017-12-09:
The initialization clustering seems to keep expanding the set of detected spikes. Either you have a very rich recording, or some spike threshold settings are set too low. Increase the magnitude of the threshold for example with 

ops.spkTh           = -6;      % spike threshold in standard deviations (4)		

or the options below. Alternatively, try just turning off the initialization and see if you get significantly worse results. Usually you won't see a big difference.

danielnee2, 2017-12-12:
Hi Marius:
Thanks so much for the suggestions. 
I tried the turn off  initialization method and it works fine without obvious detrimental effect on spike detection.
The recording is very rich indeed and may be the reason that crash the initialization clustering.
Would be nice if later kilosort versions can adapt to high firing rate recordings as well.

closed 2017-12-12 by danielnee2